sandeepmistry/node-sensortag #31: How to read other characteristics?

Nesh108, 2015-02-06:
Hello,

I am using this (awesome) library to connect my BeagleBone to a TI CC2541.

I noticed that it is possible to access most of the characteristics but I was wondering if I could read (and write) data from other services.

Namely, I would like to be able to access the Test Service and its characteristics. I have their UUID's, is there any way to use this lib for that?

Cheers and thanks a lot for your work :+1:

Nesh108, 2015-02-07:
I found that the UUID's are all set in the index.js.

It shouldn't be too hard to add a new characteristic.

Nesh108, 2015-02-11:
Hello,

I tried to access the data from the Test Config field but I was unable to read its value.
Is 'data.readInt8(0)' enough?

sandeepmistry, 2015-02-12:
@Nesh108 can you provide a larger code snippet for what you tried?

Which UUID are you trying to read - AA62? http://processors.wiki.ti.com/index.php/SensorTag_User_Guide#Test_Service

```javascript
var TEST_CONFIGURATION_UUID = 'f000aa6204514000b000000000000000';

SensorTag.prototype.readTestConfiguration = function(callback) {
  this.readDataCharacteristic(TEST_CONFIGURATION_UUID, function(data) {
    callback(data.readInt8(0));
  }.bind(this));
};
```

Pull request for new API would be great once you get it working ...

Nesh108, 2015-02-12:
Hello Sandeep,

Here is a larger snippet:

    var LOCK_CONFIG_UUID                        =     'f000aa6204514000b000000000000000';
    
    SensorTag.prototype.readLockState = function(callback) { 
      this.readDataCharacteristic(LOCK_CONFIG_UUID, function(data) {
        this.decodeLock(data, callback);
      }.bind(this));
    };
    
    SensorTag.prototype.decodeLock = function(data, callback) {
    	var locked = data.readUInt8(0);
    	callback(locked);
    };

sandeepmistry, 2015-02-13:
@Nesh108 see 9973e5a5c09a2a392ce9604620abf428228bd5f0 - it works for me.

closed 2015-02-13 by sandeepmistry

Nesh108, 2015-02-13:
Mmm weird. Which FW are you using? 1.4 or 1.5?

sandeepmistry, 2015-02-13:
1.5 (Oct 23 2013)

Nesh108, 2015-02-13:
I will try with my other sensortag and see if anything changes...

sandeepmistry, 2015-02-13:
What happens when you run the latest test.js from master?

Debug mode: ```sudo DEBUG=l2cap-ble node test.js```

Nesh108, 2015-02-14:
I think the problem was the device. With my other one, it worked flawlessly.

Thanks, dude!

sandeepmistry, 2015-02-14:
@Nesh108 no problem, glad you got it working with the other device.