sandeepmistry/node-sensortag #33: promisify

schnippe, 2015-02-14:
Hi Sandeep, 

would you accept a pull request for compatibility to promises and harmony generators? This would help out of the callback hell and look pretty nice. Harmony Generators are supported in node 0.12
I am thinking about making a Promise Wrapper around the node-sensortag API or forking it. I will retain node 0.10 compatibility.

Test.js could look like this or even shorter:

var Promise = require('bluebird');
    var SensorTag = require('sensortag')
    
    Promise.coroutine(function* () {
      var sensorTag = yield SensorTag.discover()
    
      sensorTag.on('disconnect', function () {
        console.log('disconnected!');
      });
    
      console.log('connecting...');
      yield sensorTag.connect();
    
      console.log('discoverServicesAndCharacteristics');
      yield sensorTag.discoverServicesAndCharacteristics();
    
      console.log('readSystemId');
      console.log('\tsystem id = ' + yield sensorTag.readSystemId());
    
      console.log('readSerialNumber');
      console.log('\tserial number = ' + yield sensorTag.readSerialNumber());
    
      console.log('readFirmwareRevision');
      console.log('\tfirmware revision = ' + yield sensorTag.readFirmwareRevision());
    
      console.log('readHardwareRevision');
      console.log('\thardware revision = ' + yield sensorTag.readHardwareRevision());
    
      console.log('readSoftwareRevision');
      console.log('\tsoftware revision = ' + yield sensorTag.readHardwareRevision)();
    
      console.log('readManufacturerName');
      console.log('\tmanufacturer name = ' + yield sensorTag.readManufacturerName());
    
      console.log('enableIrTemperature');
      yield sensorTag.enableIrTemperature()
      yield Promise.delay(2000);
      console.log('readIrTemperature');
      var irTemperature = yield sensorTag.readIrTemperature();
      console.log('\tobject temperature = %d °C', irTemperature.objectTemperature.toFixed(1));
      console.log('\tambient temperature = %d °C', irTemperature.ambientTemperature.toFixed(1));
    
      console.log('disableAccelerometer');
      yield sensorTag.disableAccelerometer();
      console.log('enableAccelerometer');
      yield sensorTag.enableAccelerometer();
      yield Promise.delay(2000);
      console.log('readAccelerometer');
      var accelerometer = yield sensorTag.readAccelerometer();
      console.log('\tx = %d G', accelerometer.x.toFixed(1));
      console.log('\ty = %d G', accelerometer.y.toFixed(1));
      console.log('\tz = %d G', accelerometer.z.toFixed(1));
      console.log('disableAccelerometer');
      yield sensorTag.disableAccelerometer();
    
      console.log('enableHumidity');
      yield sensorTag.enableHumidity();
      yield Promise.delay(2000);
      console.log('readHumidity');
      var humidity = yield sensorTag.readHumidity();
      console.log('\ttemperature = %d °C', humidity.temperature.toFixed(1));
      console.log('\thumidity = %d %', humidity.humidity.toFixed(1));
      console.log('disableHumidity');
      yield sensorTag.disableHumidity();
    
      console.log('enableMagnetometer');
      sensorTag.enableMagnetometer();
      yield Promise.delay(2000);
      console.log('readMagnetometer');
      var magnetometer = yield sensorTag.readMagnetometer();
      console.log('\tx = %d μT', magnetometer.x.toFixed(1));
      console.log('\ty = %d μT', magnetometer.y.toFixed(1));
      console.log('\tz = %d μT', magnetometer.z.toFixed(1));
      console.log('disableMagnetometer');
      yield sensorTag.disableMagnetometer();
    
      console.log('enableBarometricPressure');
      yield sensorTag.enableBarometricPressure();
      yield Promise.delay(2000);
      console.log('readBarometricPressure');
      console.log('\tpressure = %d mBar',
        (yield sensorTag.readBarometricPressure()).toFixed(1));
      console.log('disableBarometricPressure');
      yield sensorTag.disableBarometricPressure();
    
      console.log('enableGyroscope');
      yield sensorTag.enableGyroscope();
      yield Promise.delay(2000);
      console.log('readGyroscope');
      var gyroscope = yield sensorTag.readGyroscope();
      console.log('\tx = %d °/s', gyroscope.x.toFixed(1));
      console.log('\ty = %d °/s', gyroscope.y.toFixed(1));
      console.log('\tz = %d °/s', gyroscope.z.toFixed(1));
      console.log('disableGyroscope');
      yield sensorTag.disableGyroscope();
    
      console.log('readTestData');
      console.log('\tdata = ' + yield sensorTag.readTestData());
    
      console.log('readTestConfiguration');
      console.log('\tconfiguration = ' + yield sensorTag.readTestConfiguration());
    
      console.log('readSimpleRead, press both buttons on your SensorTag');
    
      yield new Promise(function (resolve, reject) {
        sensorTag.on('simpleKeyChange', function (left, right) {
          console.log('left: ' + left);
          console.log('right: ' + right);
          if (left && right) {
            resolve(true);
          }
        });
      });
    
      return sensorTag;
    })().
      then(function (sensorTag) {
        console.log('done testing ' + sensorTag.uuid)
      }, function (err) {
        console.error(err.stack)
      })

sandeepmistry, 2015-02-16:
@schnippe would this force users into using promises? If so, creating a "promisified" version that wraps the currently library makes more sense.

schnippe, 2015-02-16:
@sandeepmistry No, it would not force users into using promises. But it would enforce using a single api style like 

    SensorTag.discover (param1, param2, callback) // with callback(error, result1, result2)   

It would possibly mean maintaing a part of the api-signature twice, too. 

Are you interested?

sandeepmistry, 2015-03-25:
@schnippe error parameters for callbacks have been added to master.

sandeepmistry, 2015-04-07:
@schnippe any progress on this, now that callback have error parameters?

sandeepmistry, 2015-04-12:
Closing for now due to lack of activity/interest.

closed 2015-04-12 by sandeepmistry