sandeepmistry/node-sensortag #34: support for new version of Sensortag

vielmetti, 2015-03-05:
TI has a new version of the sensortag out, based on their CC2650 MCU. The design is said to have support for more sensors, including an interface to external sensors. 

http://www.ti.com/ww/en/wireless_connectivity/sensortag/index.shtml?INTC=SensorTag&HQS=sensortag

Don't have one of these in my hands yet, but hoping to track with this issue anything necessary to support it.

sandeepmistry, 2015-03-06:
@vielmetti thanks for the heads up, do you have one on order?

Have you found the equivalent page of this for it? http://processors.wiki.ti.com/index.php/SensorTag_User_Guide

vielmetti, 2015-03-06:
We have a couple on order. I looked on the TI site for any new specific info but didn't see anything particular to it, but I know at least something has changed because there is a new Android app out.

vielmetti, 2015-03-12:
Got one, started playing with it. Here's what I have found so far.

I was able to get "test.js" to recognize the device by doing the following patch:

```
Edwards-MacBook-Air:sensortag emv$ diff index.js index.js.orig
83c83
<         if ((peripheral.advertisement.localName === 'SensorTag' || peripheral.advertisement.localName === 'TI BLE Sensor Tag' || peripheral.advertisement.localName === 'CC2650 SensorTag') &&
---
>         if ((peripheral.advertisement.localName === 'SensorTag' || peripheral.advertisement.localName === 'TI BLE Sensor Tag') &&
```

That got it to successfully recognize the tag. 

This gist has a log of what it could read:  https://gist.github.com/a7efb7e98154c01698f0

This gist has a log of the error message: https://gist.github.com/43f9dcff45e88699823b

which in part reads

```
/usr/local/lib/node_modules/node-red/node_modules/sensortag/index.js:141
  this._characteristics[uuid].write(data, false, callback);
                              ^
TypeError: Cannot call method 'write' of undefined
    at SensorTag.writeCharacteristic (/usr/local/lib/node_modules/node-red/node_modules/sensortag/index.js:141:31)
    at SensorTag.disableConfigCharacteristic (/usr/local/lib/node_modules/node-red/node_modules/sensortag/index.js:175:8)
    at SensorTag.disableAccelerometer (/usr/local/lib/node_modules/node-red/node_modules/sensortag/index.js:296:8)
```

Also for completeness here's the log from the test.js program in the noble distribution

https://gist.github.com/284b8cb4783c492c1262

so progress, not complete but I think I can sort out the new UUIDs of the new functions.

vielmetti, 2015-03-12:
One more, a gist of the output of "peripheral-explorer.js":

https://gist.github.com/8c427ae160ab64c3123c

vielmetti, 2015-03-12:
Aha, and it looks like there's code for TI's Android app that supports the new sensors here:

https://git.ti.com/sensortag-android/sensortag-android

with this flle in particular showing the UUIDs used:

https://git.ti.com/sensortag-android/sensortag-android/blobs/master/src/com/example/ti/ble/sensortag/SensorTagGatt.java

and the code for scaling the sensors:

https://git.ti.com/sensortag-android/sensortag-android/blobs/master/src/com/example/ti/ble/sensortag/Sensor.java

vielmetti, 2015-03-12:
One more note to self reading through the code, the interpretation of barometric pressure has changed from the v1 sensortag to the v2 sensortag, the v2 device returns the values directly the v1 device required that you scale them. the code will need to handle that case.

vielmetti, 2015-03-13:
As part of the log of progress: Iain Hunter of TI sent some patches relative to 0.1.9 to add support for the Sensortag 2. It doesn't yet make it through all of the tests in test.js, but when it does I'm hopeful that we can merge in all of the changes and then sort out porting that all forward to add any changes from 0.1.9 to 0.1.11.

vielmetti, 2015-03-17:
Iain from TI sent two more patches by email, his report is

> I’ve now got the test.js running with cc2650 with pcb v1.20 which is the production version. Some of the data reads return 0 which still needs investigation.

Here's my results of running the newly patched test.js:

https://gist.github.com/dbe0d4bbe89ae33a7b45

Of note:

```
enableBarometricPressure
readBarometricPressure
readData len:4
convertBarometerBMP280Data: 28221,51575
        pressure = 23.3 mBar
disableBarometricPressure
enableLux
readLux
readData len:2
        Lux = 0 
disableLux
```

by which I suspect that the lux sensor and barometer are not turning up good data. (They both work from the Android phone). So progress! Next iteration will look to get these loaded under node-red.

vielmetti, 2015-03-18:
This drawing identifies all of the parts on the TI Sensortag 2 circuit board:

http://www.ti.com/lit/df/swrr134/swrr134.pdf

Of interest that is not supported yet includes

* Digital microphone SPH0641LU4H
* Buzzer HCS0503B
* Battery voltage (to detect/sense low battery conditions)
* RSSI signal strength indicator

sandeepmistry, 2015-03-21:
@vielmetti I've started a [cc2650](https://github.com/sandeepmistry/node-sensortag/tree/cc2650) branch.

I've rewrote the core to use [the 'error-parameter-callbacks' branch of noble-device](https://github.com/sandeepmistry/noble-device/tree/error-parameter-callbacks).

laurentsebag, 2015-03-21:
Hi @vielmetti , thanks for the great work. Looking forward to using node-sensortag with the cc2650. I was looking at the ti app and digging into the app, I found a list of Gatt UUIDs, not sure which ones are for the cc2650 though, I'll be playing with the code.

```
.class public Lcom/ti/ble/sensortag/SensorTagGatt;
.super Ljava/lang/Object;
.source "SensorTagGatt.java"

# static fields
.field public static final UUID_ACC_CONF:Ljava/util/UUID;

.field public static final UUID_ACC_DATA:Ljava/util/UUID;

.field public static final UUID_ACC_PERI:Ljava/util/UUID;

.field public static final UUID_ACC_SERV:Ljava/util/UUID;

.field public static final UUID_BAR_CALI:Ljava/util/UUID;

.field public static final UUID_BAR_CONF:Ljava/util/UUID;

.field public static final UUID_BAR_DATA:Ljava/util/UUID;

.field public static final UUID_BAR_PERI:Ljava/util/UUID;

.field public static final UUID_BAR_SERV:Ljava/util/UUID;

.field public static final UUID_DEVINFO_FWREV:Ljava/util/UUID;

.field public static final UUID_DEVINFO_SERV:Ljava/util/UUID;

.field public static final UUID_GYR_CONF:Ljava/util/UUID;

.field public static final UUID_GYR_DATA:Ljava/util/UUID;

.field public static final UUID_GYR_PERI:Ljava/util/UUID;

.field public static final UUID_GYR_SERV:Ljava/util/UUID;

.field public static final UUID_HUM_CONF:Ljava/util/UUID;

.field public static final UUID_HUM_DATA:Ljava/util/UUID;

.field public static final UUID_HUM_PERI:Ljava/util/UUID;

.field public static final UUID_HUM_SERV:Ljava/util/UUID;

.field public static final UUID_IRT_CONF:Ljava/util/UUID;

.field public static final UUID_IRT_DATA:Ljava/util/UUID;

.field public static final UUID_IRT_PERI:Ljava/util/UUID;

.field public static final UUID_IRT_SERV:Ljava/util/UUID;

.field public static final UUID_KEY_DATA:Ljava/util/UUID;

.field public static final UUID_KEY_SERV:Ljava/util/UUID;

.field public static final UUID_MAG_CONF:Ljava/util/UUID;

.field public static final UUID_MAG_DATA:Ljava/util/UUID;

.field public static final UUID_MAG_PERI:Ljava/util/UUID;

.field public static final UUID_MAG_SERV:Ljava/util/UUID;

.field public static final UUID_MOV_CONF:Ljava/util/UUID;

.field public static final UUID_MOV_DATA:Ljava/util/UUID;

[... 338 lines cut ...]

    sput-object v0, Lcom/ti/ble/sensortag/SensorTagGatt;->UUID_KEY_SERV:Ljava/util/UUID;

    .line 111
    const-string v0, "0000ffe1-0000-1000-8000-00805f9b34fb"

    invoke-static {v0}, Ljava/util/UUID;->fromString(Ljava/lang/String;)Ljava/util/UUID;

    move-result-object v0

    sput-object v0, Lcom/ti/ble/sensortag/SensorTagGatt;->UUID_KEY_DATA:Ljava/util/UUID;

    return-void
.end method

.method public constructor <init>()V
    .locals 0

    .prologue
    .line 61
    invoke-direct {p0}, Ljava/lang/Object;-><init>()V

    return-void
.end method
```

sandeepmistry, 2015-03-25:
@vielmetti @laurentsebag I've merge the cc2650 branch into master, you can try out via ```npm install sandeepmistry/node-sensortag```

Working on updating the read me and test.js before publishing to npm now ....

laurentsebag, 2015-03-25:
Hi @sandeepmistry, great! Thanks for letting us know. I had tried the cc2650 branch and had no issues (although I hadn't tried the gyroscope). I'll let you know if I have any issues with the master branch.

sandeepmistry, 2015-03-26:
@laurentsebag no problem. I think the only remaining issue is magnetometer notifications ...

AlexKoe, 2015-03-26:
Sorry to bother you, but I just installed (via npm -g install sandeepmistry/node-sensortag)  the most recent sensortag version which supports the CC2650 (1.0.0) on my RPi and RPi2 but unfortunately it doesn't show up in the node-red palette. What am I missing? Thx in advance!

sandeepmistry, 2015-03-28:
@AlexKoe I haven't used node-red myself, maybe @vielmetti can help you out with this.

It think you should probably discuss it somewhere more "node-red + sensor tag specific".

vielmetti, 2015-03-28:
Hi @AlexKoe, I got this far about 10 days ago

https://github.com/node-red/node-red-nodes/compare/master...vielmetti:master

in doing the tiny bit of additional work to get the node-red-node-sensortag code supported. There may be a thing or two left to do from the other changes that @sandeepmistry made.

This issue https://github.com/node-red/node-red-nodes/issues/107 is probably a better place to track it than here.

sandeepmistry, 2015-04-11:
v1.0.0 has been published to npm.

closed 2015-04-11 by sandeepmistry

vielmetti, 2015-04-21:
Cross-tracking this for a port of this work to Python, see 'bluepy' at https://github.com/IanHarvey/bluepy/issues/36

arunn-tp, 2015-05-12:
Hi Guys.

Im relatively new to TI platform and got an opportunity to develop a custom Bluetooth application using CC2650DK

I have got my hands on two CC2650 Development Kit which involves two CC2650 Evaluation Modules, two SmartRF06 Boards. 

We are using the default Bluetooth-stack2 on Code composer studio IDE 
My requirement is to get the Rssi of both the bluetooth module on LCD without using the existing menu based application which is preloaded 
Please help me to get around and understand the code and do the necessary modifications to complete my requirements

sandeepmistry, 2015-05-12:
@arunn-tp I don't think this is the right place to ask your question, as it seems unrelated to the issue, node-sensortag and noble.

Impatience71, 2016-11-23:
Hi guys, 
I am lost in the installation of sensortag on a Raspi3. 
Installed Node-Red v0.15.2
Node.js v7.2.0
When I start node-red I get a warning: 
[warn] [sensorTag] Error: Cannot find module 'bluetooth-hci-socket'

Later on in the node red desk I can place the widget but either the "Rescan" button is missing or the widget can not connect to the sensortag. (just simple flow with two widgets: SensorTag and debug).

I can connect the sensortag with gattool and can also write and read values. 

I donwloaded bluetooth-hci-socket with git clone but when I try to built with "npm install" I get a 
node-pre-gyp  ERR! Tried to download(404): https//github.com/sandeepmistry/node-bluetooth....binding-0.5.2-node-v51-linux-arm.tar.gz

when he start compiling with the CXX he is dying in the first source. 

Any help is appreciated.

Thanks in advance
Peter

sandeepmistry, 2016-11-26:
@Impatience71 your comment above is not related to the original issue.

Please open a new issue with more details, including the full output from the npm install command.