sandeepmistry/node-sensortag #37: Trying to connect to 2 tags at once

hardillb, 2015-04-09:
As mentioned over in the Node-Red side of things, I'm having issues connecting to multiple tags at once.

The second discover never happens.  I'm wondering if the discovery phase is shutdown by the connect call. If so I guess the only way to do it is to call discover all, save the list of discovered tags and then connect after a suitable time out period?

```javascript
var SensorTag = require('sensortag');

SensorTag.discoverByUuid("34b1f7d136b9",function(tag){
	console.log("found " + tag.uuid);
	tag.connect(function(){
		tag.discoverServicesAndCharacteristics(function(){
			tag.notifySimpleKey(function(err){
				console.log(err);
			});
			tag.on('simpleKeyChange', function(left, right){
				console.log("button " + left + " 34b");
			});
		})
	});
	
	
});

SensorTag.discoverByUuid('1cba8c20caf3',function(tag){
	console.log("found " + tag.uuid);
	tag.connect(function(){
		tag.discoverServicesAndCharacteristics(function(){
			tag.notifySimpleKey(function(err){
				console.log(err);
			});
			tag.on('simpleKeyChange', function(left, right){
				console.log("button " + left + " 1cb");
			});
		})
	   });
});
```

sandeepmistry, 2015-04-10:
@hardillb I can't seem to reproduce locally with 2 CC2650 sensor tags (I only own one CC2540 SensorTag, so can't test with two of those). Tried on my Mac and Raspberry Pi.

As well, a slightly modified version:

```javascript
var SensorTag = require('sensortag');

var UUID_1 = '68c90b06bc81';
var UUID_2 = '68c90b044580';

SensorTag.discoverByUuid(UUID_1, function(tag1) {
    console.log('1: found ->' + tag1.uuid);

    tag1.connectAndSetUp(function() {
        console.log('1 -> connect and set up');

        tag1.notifySimpleKey(function(err) {
            console.log('1: notifySimpleKey' + (err ? err : 'success'));
        });
    });

    tag1.on('simpleKeyChange', function(left, right) {
        console.log('1: button ->' + right);
    });

    tag1.once('disconnect', function() {
        console.log('1: disconnected');
    });
});

SensorTag.discoverByUuid(UUID_2, function(tag2) {
    console.log('2: found ->' + tag2.uuid);

    tag2.connectAndSetUp(function() {
        console.log('2: connect and set up');

        tag2.notifySimpleKey(function(err) {
            console.log('2: notifySimpleKey' + (err ? err : 'success'));
        });
    });

    tag2.on('simpleKeyChange', function(left, right) {
        console.log('2: button ->' + right);
    });

    tag2.once('disconnect', function() {
        console.log('2: disconnected');
    });
});
```

Could you please provide debug logs: ```DEBUG=* node <files>.js```

hardillb, 2015-04-10:
I still get the same with your code and 2 version 1 SensorTags.

The debug log is here: https://gist.github.com/hardillb/d17331c148128334eae0

sandeepmistry, 2015-04-10:
@hardillb any change in behaviour if you swap uuid's/BT address order?

The debug log only shows one discover event. To get more info, you could add some log statements in noble's ```startScanning``` and ```stopScanning``` methods, to verify stop scanning isn't been called.

hardillb, 2015-04-13:
Changing the order doesn't change anything (apart from which tag gets to connect, it's always the first one).

I'll try and find some time later to have a poke lower down the stack

sandeepmistry, 2015-04-26:
@hardillb any progress on this?

I'm wondering if something in the Bluetooth stack (hardware, driver, kernel version) doesn't allow for scanning while connected. What adapter and kernel version are you using?

hardillb, 2015-04-26:
Sorry, not been in the office long enough to play with this again recently. I'll try and have another look this week.

hardillb, 2015-04-29:
This is all running on Fedora Linux 20 with the following 

kernel: Linux bagend 3.19.5-100.fc20.x86_64 #1 SMP Mon Apr 20 19:51:16 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

bluez: 5.18

hardware: Broadcom Corp. BCM20702 Bluetooth 4.0

not had chance to add anything to the code yet

sandeepmistry, 2015-04-29:
@hardillb thanks, one thing to try is scan only (no connecting) - something like:

```javascript
var SensorTag = require('sensortag');

SensorTag.discoverByUuid("34b1f7d136b9",function(tag){
    console.log("found " + tag.uuid);
});

SensorTag.discoverByUuid('1cba8c20caf3',function(tag){
    console.log("found " + tag.uuid);
});
```

sandeepmistry, 2015-06-19:
@hardillb ping ... any progress?

piekstra, 2015-06-19:
I am having similar issues trying to connect to a list of CC2650 sensortags at the same time:
Here is a code-snippet (not actually tested but quite similar to my implementation)

```
var async = require('async');

var devices = [ "68c90b06a104",
                "68c90b043a01",
                "68c90b043a02",
                "68c90b043a03",
                "68c90b043a04"]
                
var asynchronous = true;
var SensorTag = require('./index');
      
if (asynchronous) { 
    async.forEach(devices, function(deviceMac, callback) {
      //directly connect to device
	  SensorTag.discoverByUuid(deviceMac, function(err) {
        console.log("found " + deviceMac); 
        callback();
      });  
    }, function (err) {
      console.log('All found!');
      process.exit(0);
    }); 
} 
```

hardillb, 2015-06-22:
@sandeepmistry Sorry got side tracked with various things.

I've run the code you suggested and it's discovering both:

    $ sudo node test.js
    [sudo] password for hardillb: 
    found 34b1f7d136b9
    found 1cba8c20caf3`

So it looks like it's the connection part

sandeepmistry, 2015-06-24:
@hardillb cool, so next step would be trying to connect to both after both are discovered ... please try that out and report back.

hardillb, 2015-06-25:
The following code seams to be working:

    var SensorTag = require('sensortag');
    var tags = [];

    SensorTag.discoverByUuid("34b1f7d136b9",function(tag){
        console.log("found " + tag.uuid);
        tags.push(tag);
    });

    SensorTag.discoverByUuid('1cba8c20caf3',function(tag){
        console.log("found " + tag.uuid);
        tags.push(tag);
    });

    setTimeout(function(){
       console.log("timer");
       tags.forEach(function(tag){
          tag.connectAndSetUp(function(){
              console.log("connected " + tag);

             tag.notifySimpleKey(function(err) {
                console.log('1: notifySimpleKey' + (err ? err : 'success'));
             });

         });
         tag.on('simpleKeyChange',function(left,right){
            console.log(left + " - " + right);
         });
       })
    },5000);

martin-doyle, 2015-06-26:
I am having a similar issue. 
I have a setup with 3 SensorTags 2650 and once in 3 starts one of the SensorTags is not discovered. What I found is that the discovery with noble and noble-device tests work fine but the peripheral.advertisement.localName is sometimes "undefined".

Then the function "is" does not recognize the device as a SensorTag:
CC2650SensorTag.is = function(peripheral) {
  var localName = peripheral.advertisement.localName;

  return (localName === 'CC2650 SensorTag') ||
          (localName === 'SensorTag 2.0');
};

martin-doyle, 2015-06-27:
I wrote a test which works with my setup: 3 SensorTags 2650 on a Raspberry via a USB Bluetooth dongle.

I set the parameter ```javascript SensorTag.SCAN_DUPLICATES = true; ``` in order to be able to connect even if the SensorTag does not send the advertisement data the first time. In addition I can reconnect just in case the connection is gets lost somehow.
Secondly I stop discovering until the discovered SensorTag is connected in order to make sure it is connected only once. Then I restart discovering. 

https://github.com/martin-doyle/node-sensortag/blob/master/testSensorTag.js

sandeepmistry, 2015-07-05:
@martin-doyle please open a new issue for the discovery issue, I would be happy to merge a pull request for ```SensorTag.SCAN_DUPLICATES = true;``` as well.

It would be good to get an ```hcidump``` capture from the scenario of trying to connect while scanning is taking place.

martin-doyle, 2015-07-06:
@sandeepmistry I am not sure which part you want to pull in. Do you mean the whole test?

sandeepmistry, 2015-07-07:
@martin-doyle just the ```SensorTag.SCAN_DUPLICATES = true;``` part if it fixes #45 for you.

jeremyjohnson, 2015-11-16:
Hi!  I have a related question, which may possibly stem from a misunderstanding of how BLE is meant to function. Sorry if this is the wrong place to ask this - if it is, plase can you point me at teh correct one?  Thanks!  Following on from hardlib's suggestion of a list of known sensortags,  If one wanted to connect to and disconnect from an array of sensortags of known address/ID, is the 'discovery' phase always necessary, or can one go directly to the 'connect' step?   To express it in terms of Sandeep's node.js code: in a node.js setup, is using one of the 'discover' factories the only way to instantiate a sensorTag object, or is there some kind of SensorTag.new() pattern where an ID can be assigned from a list of already-known IDs?  My issue is that in order for discoverById to succeed, the tag must be forced to advertise (or flashed with new firmware to constantly advertise in the absence of a connection, which would kill the battery)

hardillb, 2015-11-16:
I think that if the tag is not connected or advertising it is powered down so you couldn't connect to it anyway

sandeepmistry, 2015-11-20:
@jeremyjohnson your question is a bit off topic to this issue. Gitter is a better place for questions like this: https://gitter.im/sandeepmistry/node-sensortag

However, I agree with @hardillb's answer above. The discover phase is required by this library, as the BLE lib it uses ([noble](https://github.com/sandeepmistry/noble)) requires it.

ScottChapman, 2016-02-09:
I may have run into a similar situation. I am only seeing one of my two 2650's discovered with discoverAll.

Is it really discoverAny?

martin-doyle, 2016-02-09:
What I found is that discovering stops when some data is sent. This the case when you do `connectAndSetUp`.
In order to fix that I do the following:

1. Discover All
2. On connect stop discovering
3. When connected start discovering again

You can find a test script at https://github.com/martin-doyle/SensorTagMQTT/blob/master/testSensorTag.js.

I could connect up to 5 SensorTags so far. Please let me know if that works for you.

ScottChapman, 2016-02-10:
So, I;ve tried connecting by ID, address, and all and they all seem to stop once I connect to the first one.

It appears the only way to do this, as you suggest, is to stop/start discovery after each discovery. Will work to incorporate your code into what I am trying.

Thanks!

martin-doyle, 2016-02-16:
@ScottChapman Just curious. Did it work?

ScottChapman, 2016-02-16:
@martin-doyle - Yep! I was able to leverage your code for my app. Seems like the pattern could be incorporated into the node-sensortag code to effectively provide expected behavior.

Thanks!

sandeepmistry, 2016-02-21:
@ScottChapman 

>  Seems like the pattern could be incorporated into the node-sensortag code to effectively provide expected behaviour.

I can put a patch upstream in [noble](https://github.com/sandeepmistry/noble) to fix this, would you have time to test if I do this?

ScottChapman, 2016-02-21:
Sure, maybe not right away but would be happy to noodle on it.

martin-doyle, 2016-02-23:
@sandeepmistry I would test the patch, too.

mcleod-ic, 2016-03-19:
Could my problem be related to having two 2541 SensorTags?

I have two SensorTags at home and the SensorTag node shows up in NodeRed but never loads anything. As soon as one tag is connected, node-red crashes. Using Edison with node-red 0.9 and node 0.10 (currently part-way through a remake of node 0.10 to get it to 0.12, as noble seems to want this) 

I do see data from the SensorTags on other applications, so I was wondering if the presence of the two together might be resulting in 0 payloads.

I have used hci-tool to scan, pair and trust each of the tags, and they do show up in bluetoothctl

sandeepmistry, 2016-03-28:
@mcleod-ic I would suggest you create an issue here: https://github.com/node-red/node-red-nodes

danielbh, 2016-04-21:
@martin-doyle Thanks for the advice. It's my best option at the moment. @sandeepmistry  is there any news on if this is fixable? This bug is really hurting the quality of my code. I'd be happy to help test or fix things. Just not sure where to start.

dbrendon, 2016-05-16:
@sandeepmistry Was there ever a patch in noble for handling multiple devices (specifically CC2650)?

stefanluyten, 2016-06-10:
@sandeepmistry Hi - is there a noble patch for handling multiple CC2650?

sandeepmistry, 2016-06-20:
@danielbh @danielbh @stefanluyten unfortunately I haven't looked at a patch for the noble side yet.

danielbh, 2016-07-17:
This is no longer an issue for me. I realized as long as I do good control flow and dont connect while discovering it works great. Thanks for your hard work!

danielbh, 2016-07-17:
By the way I connect 8 tags at one time. I found it to be the hard limit, but I think you say that in the docs somewhere.

sandeepmistry, 2016-07-18:
@danielbh cool, I'm up for adding a note about "good flow control" and maximum to the read me. Please start a pull request to get the discussion going. The maximum is a bit tricky, as it's hardware/firmware and OS dependent (https://github.com/sandeepmistry/noble#maximum-simultaneous-connections).

saurabh-prophecy, 2017-02-12:
@sandeepmistry : i am connecting 3-4 sensortags..first time they all  got connected but after restarting /rebooting my rpi few of them got disconnected. I think while connecting some of the sensortags they got stuck inside connectAndSetup. what should i do to resolve this???

sandeepmistry, 2017-02-21:
@saurabh-prophecy your issue seems unrelated to the original topic, please open a new issue with more details for your specific problem.

inanbayram, 2017-07-03:
@danielbh how did you connect 8 tags at one time? After the first tag, I cannot connect the second one :(

inanbayram, 2017-07-03:
@inanbayram

martin-doyle, 2017-07-14:
Look at the comments about flow control. (9 Feb 2016 and the following)

mukutprasad1992, 2019-04-24:
Hi, I have two noble listeners one is listening only and another one is scan and connect. If Noble is scanning and then connect after that 1 connection attempts never automatically start scanning. I have to start scanning manually "Noble.startScanning()" after every broadcast. Is there any solution to prevent them. Thanks in advance.