sandeepmistry/node-sensortag #44: Sensortag drops/rejects connections

piekstra, 2015-07-02:
I am having an issue where the tag drops/rejects connections some times. The device is discovered successfully and I get to the point of connectAndSetUp but quickly after that the device disconnects and the .on('disconnect', ...) is triggered in the js code. 

This issue has been repeated with both CSR and Broadcom chips. It also occurs with both a python version of the code and this node.js version of the code.

Verbose log output of one such failed connection:
[06-30 23:58:02 UTC] preparing to discoverByUuid 0123456789ab 
[06-30 23:48:05 UTC] NOBLE onDiscover 0123456789ab 01:23:45:67:89:ab public { localName: 'CC2650 SensorTag',
  txPowerLevel: 0,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ 'aa10' ] } -70
[06-30 23:48:05 UTC] discovered: {"uuid":"0123456789ab","type":"cc2650"}
[06-30 23:48:05 UTC] connected 0123456789ab
[06-30 23:48:05 UTC] connectAndSetUp
[06-30 23:48:05 UTC] NOBLE connect 0123456789ab
[06-30 23:48:05 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "info using 5C:F3:70:68:C3:FF@hci0\n"
[06-30 23:48:05 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = info using 5C:F3:70:68:C3:FF@hci0
[06-30 23:48:08 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "connect success\n"
[06-30 23:48:08 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = connect success
[06-30 23:48:08 UTC] l2cap-ble onStdoutData connect 01:23:45:67:89:ab null
[06-30 23:48:08 UTC] NOBLE onConnect 0123456789ab null
[06-30 23:48:08 UTC] NOBLE discoverServices 0123456789ab []
[06-30 23:48:08 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "write = success\n"
[06-30 23:48:08 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = write = success
[06-30 23:48:09 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "disconnect\n"
[06-30 23:48:09 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = disconnect
[06-30 23:48:09 UTC] l2cap-ble onStdoutData disconnect 01:23:45:67:89:ab
[06-30 23:48:09 UTC] BINDINGS onDisconnect 01:23:45:67:89:ab 0123456789ab
[06-30 23:48:09 UTC] NOBLE onDisconnect 0123456789ab
[06-30 23:48:09 UTC] disconnected!

On a successful connection, after the line:
[06-30 23:48:08 UTC] NOBLE discoverServices 0123456789ab []

The following is logged:
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "write = success\n"
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = write = success
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "data 031700\n"
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = data 031700
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "write = success\n"
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = write = success
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "data 110601000700001808000b0001180c001e000a18\n"
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = data 110601000700001808000b0001180c001e000a18
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "write = success\n"
[06-30 23:58:06 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = write = success
[06-30 23:58:07 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "data 11141f00260000000000000000b00040510400aa00f0\n"
[06-30 23:58:07 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = data 11141f00260000000000000000b00040510400aa00f0
[06-30 23:58:07 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "write = success\n"
...
lots more

sandeepmistry, 2015-07-05:
@piekstra please share an ```sudo hcidump``` capture, it might provide some more additional info.

It might be a firmware/device issue, of something with the Linux kernel. Especially since it happens with noble as well as Pything.

piekstra, 2015-07-06:
Here is the log with hcidump and additional logging:

> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00

[07-06 17:42:52 UTC] preparing to discoverByUuid 0123456789ab
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00

[07-06 17:42:52 UTC] NOBLE onDiscover 0123456789ab 01:23:45:67:89:ab public { localName: 'CC2650 SensorTag',
  txPowerLevel: 0,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ 'aa10' ] } -59
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00

[07-06 17:42:52 UTC] discovered: {"uuid":"0123456789ab","type":"cc2650"}
[07-06 17:42:52 UTC] connected 0123456789ab
[07-06 17:42:52 UTC] connectAndSetUp
[07-06 17:42:52 UTC] NOBLE connect 0123456789ab
[07-06 17:42:52 UTC] l2cap-ble connect 01:23:45:67:89:ab: l2capBle = /home/pi/node_modules/sensortag/node_modules/noble-device/node_modules/noble/lib/linux/../../build/Release/l2cap-ble
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1

[07-06 17:42:52 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "info using 5C:F3:70:68:C3:FF@hci0\n"
[07-06 17:42:52 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = info using 5C:F3:70:68:C3:FF@hci0
[07-06 17:42:53 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "connect success\n"
[07-06 17:42:53 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = connect success
[07-06 17:42:53 UTC] l2cap-ble onStdoutData connect 01:23:45:67:89:ab null
[07-06 17:42:53 UTC] l2cap-ble _queueCommand 01:23:45:67:89:ab: write:  <Buffer 02 00 01> function () { [native code] } undefined
[07-06 17:42:53 UTC] l2cap-ble _queueCommand 01:23:45:67:89:ab: write: 020001
[07-06 17:42:53 UTC] NOBLE onConnect 0123456789ab null
[07-06 17:42:53 UTC] NOBLE discoverServices 0123456789ab []
[07-06 17:42:53 UTC] l2cap-ble readByGroupRequest 1 65535 10240 <Buffer 00 00 00 00 00 00 00>
[07-06 17:42:53 UTC] l2cap-ble _queueCommand 01:23:45:67:89:ab: write:  <Buffer 10 01 00 ff ff 00 28> function () { [native code] } undefined
[07-06 17:42:53 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "write = success\n"
[07-06 17:42:53 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = write = success
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x3e
    Reason: Connection Failed to be Established

[07-06 17:42:53 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: buffer = "disconnect\n"
[07-06 17:42:53 UTC] l2cap-ble onStdoutData 01:23:45:67:89:ab: line = disconnect
[07-06 17:42:53 UTC] l2cap-ble onStdoutData disconnect 01:23:45:67:89:ab
[07-06 17:42:53 UTC] BINDINGS onDisconnect 01:23:45:67:89:ab 0123456789ab
[07-06 17:42:53 UTC] NOBLE onDisconnect 0123456789ab
[07-06 17:42:53 UTC] disconnected!

sandeepmistry, 2015-07-19:
@piekstra thanks for the logs, I'm putting this one on hold. Will revisit once the ```bluetooth-hci-socket``` branch is integrated in noble: https://github.com/sandeepmistry/noble/tree/bluetooth-hci-socket

sandeepmistry, 2015-09-20:
@piekstra please try again with the latest version.

sandeepmistry, 2015-09-28:
@piekstra any updates?

piekstra, 2015-09-28:
Sorry!
@sandeepmistry I am unable to do any further tests as I no longer have access to the project I was working on that implemented your Node.js code for communicating with the sensortags.

sandeepmistry, 2015-09-28:
@piekstra no worries and thanks for letting me know. I'll close this for now, but feel free to re-open if you get access to the project in the future.

closed 2015-09-28 by sandeepmistry