sandeepmistry/node-sensortag #45: SensorTag sometimes not dicovered

martin-doyle, 2015-07-06:
I have a setup with 2 SensorTags 2650 and once in awhile one of the SensorTags is not discovered. What I found is that the discovery with noble and noble-device tests work fine but the peripheral.advertisement.localName is sometimes "undefined".

I use a very simple test program:
```javascript
var noble = require('noble');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
    console.log('peripheral with UUID ' + peripheral + ' found');
    var advertisement = peripheral.advertisement;
});

```

In most cases a get the following result:
```
peripheral with UUID {
  "uuid": "c4be84727309",
  "address": "c4:be:84:72:73:09",
  "advertisement": {
    "localName": "CC2650 SensorTag",
    "txPowerLevel": 0,
    "serviceData": [],
    "serviceUuids": [
      "aa10"
    ]
  },
  "rssi": -65,
  "state": "disconnected"
} found
peripheral with UUID {
  "uuid": "c4be8471a68d",
  "address": "c4:be:84:71:a6:8d",
  "advertisement": {
    "localName": "CC2650 SensorTag",
    "txPowerLevel": 0,
    "serviceData": [],
    "serviceUuids": [
      "aa10"
    ]
  },
  "rssi": -61,
  "state": "disconnected"
} found
```
But sometimes the advertisemant data is incomplete:
```
peripheral with UUID {
  "uuid": "c4be84727309",
  "address": "c4:be:84:72:73:09",
  "advertisement": {
    "localName": "CC2650 SensorTag",
    "txPowerLevel": 0,
    "serviceData": [],
    "serviceUuids": [
      "aa10"
    ]
  },
  "rssi": -63,
  "state": "disconnected"
} found
peripheral with UUID {
  "uuid": "c4be8471a68d",
  "address": "c4:be:84:71:a6:8d",
  "advertisement": {
    "serviceData": [],
    "serviceUuids": [
      "aa10"
    ]
  },
  "rssi": -61,
  "state": "disconnected"
} found
```
Then the function "is" does not recognize the device as a SensorTag:
```javascript
CC2650SensorTag.is = function(peripheral) {
  var localName = peripheral.advertisement.localName;

  return (localName === 'CC2650 SensorTag') ||
          (localName === 'SensorTag 2.0');
};
```

I am not familiar with hcidump. This is what I got from starting the JS file:
```
HCI sniffer - Bluetooth packet analyzer ver 5.31
device: hci0 snap_len: 1500 filter: 0xffffffff
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```
No output during connection.
Stopping the discovery send this message:
```
HCI sniffer - Bluetooth packet analyzer ver 5.31
device: hci0 snap_len: 1500 filter: 0xffffffff
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

martin-doyle, 2015-07-06:
I wrote a test which worked with 5 SensorTags 2650 on a Raspberry via a USB Bluetooth dongle.

I set the parameter ```SensorTag.SCAN_DUPLICATES = true; ``` in order to be able to connect even if the SensorTag does not send the advertisement data the first time. In addition I can reconnect just in case the connection gets lost somehow.
Secondly I stop discovering until the discovered SensorTag is connected in order to make sure it is connected only once. Then I restart discovering. 

https://github.com/martin-doyle/node-sensortag/blob/master/testSensorTag.js

This only works if one can live with ```discoverAll```. I try to figure out how to get it running with ```discoverByUuid```.

sandeepmistry, 2015-07-19:
@martin-doyle please submit a pull request for the ```SensorTag.SCAN_DUPLICATES = true; ``` change.

martin-doyle, 2015-07-20:
@sandeepmistry I only set the parameter ```SensorTag.SCAN_DUPLICATES = true; ``` which is declared in noble-device and did not change anything in the SensorTag library.
In my program with upto 5 SensorTags I use the discovery and connect as in the test https://github.com/martin-doyle/node-sensortag/blob/master/testSensorTag.js.

sandeepmistry, 2015-08-20:
@martin-doyle do you mind trying again with the latest version of node-sensortag (without the change you suggested).

martin-doyle, 2015-08-22:
@sandeepmistry I tried the above mentioned test first. It still happens that the SensorTag does not send the localName (see third entry). By the way the second SensorTag has the new firmware 1.12.
```
peripheral with UUID {
  "uuid": "c4be84727309",
  "address": "c4:be:84:72:73:09",
  "advertisement": {
    "localName": "CC2650 SensorTag",
    "txPowerLevel": 0,
    "serviceData": [],
    "serviceUuids": [
      "aa10"
    ]
  },
  "rssi": -65,
  "state": "disconnected"
} found
peripheral with UUID {
  "uuid": "c4be84716607",
  "address": "c4:be:84:71:66:07",
  "advertisement": {
    "localName": "CC2650 SensorTag",
    "txPowerLevel": 0,
    "manufacturerData": [
      13,
      0,
      3,
      0,
      0
    ],
    "serviceData": [],
    "serviceUuids": [
      "aa80"
    ]
  },
  "rssi": -61,
  "state": "disconnected"
} found
peripheral with UUID {
  "uuid": "c4be8471a68d",
  "address": "c4:be:84:71:a6:8d",
  "advertisement": {
    "serviceData": [],
    "serviceUuids": [
      "aa10"
    ]
  },
  "rssi": -76,
  "state": "disconnected"
} found
```

sandeepmistry, 2016-09-26:
This should be solved by #86.

closed 2016-09-26 by sandeepmistry