sandeepmistry/node-sensortag #46: SensorTag CC2650 barometric pressure failing on firmware 0.89

bChiquet, 2015-07-26:
While running test.js on 0.89 firmware (USE_READ=true) :

<things work properly until we arrive to the barometric pressure>
readBarometricPressure

/home/bec/Documents/LifePlus/sensorTagNode/node-sensortag/lib/cc2650.js:86
    flTempBMP = mantissa * Math.pow(2, exponent) / 100.0;
                ^
ReferenceError: mantissa is not defined
    at CC2650SensorTag.convertBarometricPressureData (/home/bec/Documents/LifePlus/sensorTagNode/node-sensortag/lib/cc2650.js:86:17)
    at null.<anonymous> (/home/bec/Documents/LifePlus/sensorTagNode/node-sensortag/lib/common.js:151:10)
    at Characteristic.<anonymous> (/home/bec/Documents/LifePlus/sensorTagNode/node-sensortag/node_modules/noble-device/node_modules/noble/lib/characteristic.js:40:7)
    at Characteristic.g (events.js:180:16)
    at Characteristic.emit (events.js:98:17)
    at Noble.onRead (/home/bec/Documents/LifePlus/sensorTagNode/node-sensortag/node_modules/noble-device/node_modules/noble/lib/noble.js:281:20)
    at emit (events.js:106:17)
    at nobleBindings.onRead (/home/bec/Documents/LifePlus/sensorTagNode/node-sensortag/node_modules/noble-device/node_modules/noble/lib/linux/bindings.js:240:8)
    at emit (events.js:106:17)
    at null.<anonymous> (/home/bec/Documents/LifePlus/sensorTagNode/node-sensortag/node_modules/noble-device/node_modules/noble/lib/linux/l2cap-ble.js:527:14)

The barometric pressure conversion works properly with firmware 0.91

bChiquet, 2015-07-27:
Apparently the code : 
flTempBMP = mantissa * Math.pow(2, exponent) / 100.0;
should be changed to : 
flTempBMP = tempMantissa * Math.pow(2, tempExponent) / 100.0;

Will test it and commit later on.

sandeepmistry, 2015-07-30:
Fixed via #47, published to npm in version 1.0.2.

closed 2015-07-30 by sandeepmistry