sandeepmistry/node-sensortag #48: Error enabling humidity sensor on a CC2650

BastiOfBerlin, 2015-08-09:
Hi,

When enabling the humidity sensor on the CC2650, I get the error `Error: service uuid f000aa2004514000b000000000000000 not found!`
I get this immediately after calling `sensorTag.enableHumidity(callback);`

My code looks as follows:
```javascript
SensorTag.discoverByUuid(uuid, function(sensorTag) {
	console.log(sensorTag);
	sensorTag.on('disconnect', function() {
		console.log('disconnected!');
	});
	console.log('Connecting/Setting up...');
	sensorTag.connectAndSetUp(callback);
	setTimeout(function(){
		console.log('Enabling Humidity...');
		sensorTag.enableHumidity(callback);
	},1500);
});
```

Did I miss something?

BR

sandeepmistry, 2015-08-10:
@BastiOfBerlin what firmware version is your SensorTag running?

sandeepmistry, 2015-08-10:
@BastiOfBerlin also, ```connectAndSetUp``` might take longer than 1.5 seconds to complete. You could try something like this instead:

```javascript
SensorTag.discoverByUuid(uuid, function(sensorTag) {
    console.log(sensorTag);
    sensorTag.on('disconnect', function() {
        console.log('disconnected!');
    });

    console.log('Connecting/Setting up...');
    sensorTag.connectAndSetUp(function() {
        console.log('Enabling Humidity...');
        sensorTag.enableHumidity(callback);
   });
});
```

BastiOfBerlin, 2015-08-10:
Hi @sandeepmistry 
Looks like the 1.5 seconds where indeed too short, the asynchronous way fixed it for me. Thanks!

Interesting thing is: Using software, I read out the values
```
firmware revision = 0.91 (Feb 13 2015)
hardware revision = PCB 1.2
software revision = PCB 1.2
```
But on the PCB it says FW: 1.0.1... -_-

closed 2015-08-10 by BastiOfBerlin