sandeepmistry/node-sensortag #52: Not connecting to sensortags when using discoverAll

beeva-marianmoldovan, 2015-08-24:
When I use the method discoverAll, the method connectAndSetup it's not working, although the discovery it's working for all the nearby devices. When using discover all works fine.

```
var SensorTag = require('sensortag');

function onDiscover(sensorTag) {
	console.log("SensorTag discovered!");
	sensorTag.connectAndSetup(function () {
	    console.log('Connected and setup');
		sensorTag.disconnect(function(){
			console.log(sensorTag.address, 'disconnected');
		});
  	});
}

SensorTag.discoverAll(onDiscover);
```

Thanx and congrats for this awesome package

sandeepmistry, 2015-08-29:
@beeva-marianmoldovan are you using OS X or Linux? Which version? How many SensorTags?

beeva-marianmoldovan, 2015-08-31:
Linux  3.13.0-62-generic #102-Ubuntu 14.04 x86_64 GNU/Linux 

I am trying [2-3] sensors at a time

sandeepmistry, 2015-09-05:
@beeva-marianmoldovan please provide debug logs: ```sudo DEBUG=* node <file>.js```.

magicrudi, 2015-09-07:
Hi Sandeep,

first of thanks for this library. I am connecting several sensor tags with a raspberry pi and it worked until the recent update. I need to comment out line 19 in the sensortag.js file now to get it working again. Not sure if you planned it to stop discovering available sensor tags after the first one has been discovered. In the sensortag.js file:

[image]

Maybe this helps.

Rudi

sandeepmistry, 2015-09-07:
@magicrudi your comment is a bit unrelated to the original issue, see: https://github.com/sandeepmistry/node-sensortag#discover

```discover``` only discovers a single tag, ```discoverAll``` discovers more than one

magicrudi, 2015-09-07:
Sorry, my fault. I mixed up different sources.

beeva-marianmoldovan, 2015-09-08:
Didn't read nothing interesting, code here https://gist.github.com/beeva-marianmoldovan/a2982bbd7905ca114fa0

```

  hci setting filter to: 1400000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +1ms
  hci set le event mask - writing: 010120081f00000000000000 +0ms
  hci read local version - writing: 01011000 +0ms
  hci read bd addr - writing: 01091000 +0ms
  hci onSocketData: 040e0401010c00 +27ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 3073 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e0401012000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8193 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e0c0101100006bd1e06020000fc +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4097 +0ms
  hci 		status = 0 +0ms
  hci 		result = 06bd1e06020000fc +0ms
  hci set scan enabled - writing: 010c20020001 +0ms
  hci set scan parameters - writing: 010b200701100010000000 +1ms
  hci onSocketData: 040e0a01091000afd4fc87d9c4 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4105 +0ms
  hci 		status = 0 +0ms
  hci 		result = afd4fc87d9c4 +0ms
  hci address = c4:d9:87:fc:d4:af +0ms
  hci onSocketData: 040e04010c200c +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e04010b2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble stateChange poweredOn +0ms
  hci set scan enabled - writing: 010c20020101 +1ms
  hci set scan enabled - writing: 010c20020101 +0ms
  hci onSocketData: 040e04010c2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble scanStart +0ms
  hci onSocketData: 040e04010c200c +5ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +0ms
  hci onSocketData: 043e1302010000066bb848b4b007020105030210aacc +35ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +1ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 0000066bb848b4b007020105030210aacc +0ms
[... 172 lines cut ...]
  hci onSocketData: 043e2a020103014490c71917c61e0201061aff4c000215fac628359be646e09d11a7ac0292966000190001c6cc +36ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 03014490c71917c61e0201061aff4c000215fac628359be646e09d11a7ac0292966000190001c6cc +0ms
  hci 			type = 3 +0ms
  hci 			address = c6:17:19:c7:90:44 +0ms
  hci 			address type = random +0ms
  hci 			eir = 0201061aff4c000215fac628359be646e09d11a7ac0292966000190001c6 +0ms
  hci 			rssi = -52 +0ms
  gap advertisement = {"manufacturerData":[76,0,2,21,250,198,40,53,155,230,70,224,157,17,167,172,2,146,150,96,0,25,0,1,198],"serviceData":[],"serviceUuids":[]} +1ms
  hci onSocketData: 043e2a020103015e3e98249cc51e0201061aff4c000215fac628359be646e09d11a7ac0292966000190002c6cc +7ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 03015e3e98249cc51e0201061aff4c000215fac628359be646e09d11a7ac0292966000190002c6cc +0ms
  hci 			type = 3 +0ms
  hci 			address = c5:9c:24:98:3e:5e +0ms
  hci 			address type = random +0ms
  hci 			eir = 0201061aff4c000215fac628359be646e09d11a7ac0292966000190002c6 +0ms
  hci 			rssi = -52 +0ms

```

oroce, 2015-09-12:
I think I'm seeing something similar.

I tried running `DEBUG=* sudo node test.js`, none of the sensortags could be connected. After that I changed `discover` to `discoverAll` in test.js and reran the tests.

The device was discovered in both cases but couldnt be connected. It was trying to connect (`connectAndSetup` was called, and later I was seeing `noble: unkown device <uuid> connected`) but couldnt be discovered (probably?). I could successfully connect to both of the sensortags from my phone. (UUIDs of sensortags which are in range: `b4994c64be83` and `1cba8c20e2c8`)

Here are the details: https://gist.github.com/oroce/a43abb68f0bbde8b038c

sandeepmistry, 2015-09-20:
@beeva-marianmoldovan I just ran your code from https://github.com/sandeepmistry/node-sensortag/issues/52#issue-102777094 on my Raspberry Pi with a CC2540 sensor tag and CC2650 in range, both connected no problems.

sandeepmistry, 2015-09-20:
@beeva-marianmoldovan you please try running https://github.com/sandeepmistry/node-sensortag/issues/52#issue-102777094 again with a hcidump capture: ```sudo hcidump -t -x```

sandeepmistry, 2015-09-20:
@oroce thanks for the thorough gist, something is really weird, lots of ```LE Create Connection Cancel``` commands - something noble doesn't use.

 Could you please create a new issue for this?

Please confirm:
> I tried running DEBUG=* sudo node test.js, none of the sensortags could be connected. After that I changed discover to discoverAll in test.js and reran the tests.

When you create the new issue, that you can't connect to any Sensortags ...

oroce, 2015-09-27:
@sandeepmistry here's the issue: https://github.com/sandeepmistry/node-sensortag/issues/58

(on sunday the sd card got corrupted so I have to reinstall raspbian, but the issue still exists)

sandeepmistry, 2015-09-28:
@beeva-marianmoldovan ping ... https://github.com/sandeepmistry/node-sensortag/issues/52#issuecomment-141793463

beeva-marianmoldovan, 2015-09-28:
Sorry, here you have https://gist.github.com/beeva-marianmoldovan/41e3913acf4b5cfef0da

sandeepmistry, 2015-09-28:
@beeva-marianmoldovan what Bluetooth 4.0 adapter are you using?

```
2015-09-28 09:55:38.562458 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr B0:B4:48:B8:17:03 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-09-28 09:55:38.563711 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed
```

For whatever reason it doesn't like scanning and trying to connect at the same time. Also, have you tried to use a newer Linux kernel?

sandeepmistry, 2015-10-04:
@beeva-marianmoldovan any updates?

sandeepmistry, 2015-11-15:
Closing do to lack of activity.

closed 2015-11-15 by sandeepmistry