sandeepmistry/node-sensortag #53: Error Making bluetooth-hci-socket

krishnandusarkar, 2015-08-26:
While running npm update, I'm facing this error. While googling it seemed that it's Debian specific issue, but I have nodejs-legacy installed. Still the issue is not solved

```
make: Entering directory `/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
../libusb/libusb/os/threads_posix.c:24:0: warning: "_GNU_SOURCE" redefined [enabled by default]
<command-line>:0:0: note: this is the location of the previous definition
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:21: fatal error: libudev.h: No such file or directory
compilation terminated.
make: *** [Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory `/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.8.13-bone70
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding"
gyp ERR! cwd /home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:766:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:833:5)
node-pre-gyp ERR! System Linux 3.8.13-bone70
node-pre-gyp ERR! command "node" "/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
node-pre-gyp ERR! node -v v0.10.38
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing usb@1.0.6

> bluetooth-hci-socket@0.3.3 install /home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory `/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../node_modules/nan/nan.h:318:47: error: 'REPLACE_INVALID_UTF8' is not a member of 'v8::String'
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory `/home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.8.13-bone70
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/debian/SensorTag/node_modules/noble/node_modules/bluetooth-hci-socket
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.3.3
```

krishnandusarkar, 2015-08-26:
The issue can be solved by installing 

```
sudo apt-get install libudev-dev
```

closed 2015-08-26 by krishnandusarkar

rafasimionato, 2017-10-30:
Thanks. I got this same problem running Fedora 22 dist and I fixed it installing systemd-devel which made libudev.h available in my system. I used following command line :

dnf install systemd-devel