sandeepmistry/node-sensortag #54: Magnetic Switch input to cc2650

dceejay, 2015-08-27:
The cc2650 also has a magnetic switch input built in... it appears as bit 2 in the simpleswitch...
can this be exposed upwards via the simpleswitch methods... eg in common.js lines 183

  var mag = (b & 0x4) ? true : false;
  callback(left, right, mag);

sandeepmistry, 2015-08-31:
@dceejay very cool!

How would I test out triggering the magnetic switch? I just tried with a small fridge magnet, no notifications ....

dceejay, 2015-09-09:
It is located near one of the corners - must admit it wasn't very sensitive - I used a 1" round fridge type magnet or a 1cm Neodymium "super" magnet. It is the corner next to the two oval plated through holes.

sandeepmistry, 2015-09-15:
@dceejay thanks for the tip!

I've added support in https://github.com/sandeepmistry/node-sensortag/commit/59276aaebb8103fc51b9790719e0f0259455b853

__Note:__ I've called the argument ```reedRelay``` based on the TI wiki page: http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#Simple_Keys_Service

closed 2015-09-15 by sandeepmistry

dceejay, 2015-09-21:
Are you planning to bump the npm version any time soon ? (not a problem - just wondering)

sandeepmistry, 2015-09-21:
@dceejay thanks for the reminder! I've published v1.2.0 to npm with the new feature.