sandeepmistry/node-sensortag #55: Sensor Disconnected On Connection Beaglebone Black Ubuntu

snowbreeze, 2015-08-27:
Hi Sandeep,
I tried to run the test.js. It is discovering the sensor and immediately shutting down.
I have a beaglebone black, with a kinivo btd 400 usb ble 4.0. I have installed ubuntu

```
Linux arm 4.1.5-ti-r10 #1 SMP PREEMPT Thu Aug 13 20:43:10 UTC 2015 armv7l armv7l armv7l GNU/Linux
```

```
ubuntu@arm:~/sandytest/node-sensortag$ sudo node test.js 
discovered: {"id":"68c90b17cd0c","type":"cc2650"}
connectAndSetUp
readDeviceName
	device name = SensorTag 2.0
readSystemId
	system id = 68:c9:0b:00:00:17:cd:0c
readSerialNumber
	serial number = N.A.
readFirmwareRevision
	firmware revision = 1.12 (Jun 23 2015)
readHardwareRevision
	hardware revision = PCB 1.2/1.3
readSoftwareRevision
	software revision = PCB 1.2/1.3
readManufacturerName
	manufacturer name = Texas Instruments
enableIrTemperature
disconnected!
```

sandeepmistry, 2015-08-29:
@snowbreeze please provide debug logs and hcidump capture:

```sh
sudo DEBUG=* node test.js
```

```sh
sudo hcidump -t -x
```

How close is the SensorTag? I believe I was having issues with kernel 4.1 as well, kernel 4.0 worked better.

martin-doyle, 2015-09-11:
I had a similar issue after upgrading the Raspberry Pi firmware to version 4.1. The connection failed also using gatttool. 
After downgrading to version 3.18 again fixed the problem.

sandeepmistry, 2015-09-20:
@snowbreeze any updates?

snowbreeze, 2015-09-20:
I shifted to old kernel and it works fine...@krishnandu is continuing the research on the new kernel.

sandeepmistry, 2015-09-23:
@snowbreeze @martin-doyle if you still have access to something running the newer kernel, can you please check the output of ``dmesg``, I think it's related to this change: https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/commit/net/bluetooth?id=25ba265390c09b0a2b2f3fd9ba82e37248b7a371

So if you see: ```"SMP security requested but not available``` that's the issue.

tjandt, 2015-09-23:
Hi ... I'm having Kernel 4.1.6 on RaspberryPi and get this message in dmesg ... so the only solution for now is to go back to an old kernel?

sandeepmistry, 2015-09-28:
@tjandt yes, downgrading the kernel is my recommended short term work around for now, see: https://github.com/sandeepmistry/noble/issues/242

sandeepmistry, 2015-11-15:
A v1.2.1 of noble has been published with a fix for https://github.com/sandeepmistry/noble/issues/242 a fresh install of node-sensortag will pull that version down.

closed 2015-11-15 by sandeepmistry