sandeepmistry/node-sensortag #58: Cannot connect to sensortag

oroce, 2015-09-27:
This one was also mentioned [here](https://github.com/sandeepmistry/node-sensortag/issues/52#issuecomment-139747252) but @sandeepmistry asked me to open a separate issue for this.

So I cannot connect to sensortags from raspberry pi, I have 2 Sensortag CC2540, they are discovered by `node-sensortag` but connection cannot be made. (It works from my phone, i was testing with a Nexus 6 with Android 5.1.1)

Here's the gist which covers (hopefully) everything: https://gist.github.com/oroce/a43abb68f0bbde8b038c

Since I've created the report I updated my raspberry pi to latest kernel and reinstalled all the deps, so here's the updated one: https://gist.github.com/oroce/27287ae1cf36ee19cae7

And here's a really strange thing, if I run `DEBUG=* node test` the device gets disconnected in `connectAndSetup` phase while if I run `node test` without any debugging (`DEBUG=''`) the disconnect happens in `readSerialNumber` phase. Maybe there isn't any problem with noble but raspberry pi is just too slow for noble? (commands are just running out of time?)  
Btw since `noble@1.0.0` I'm seeing higher load on my raspi than I was seeing with `noble@0.3`.

sandeepmistry, 2015-09-28:
@oroce thanks for creating a new issue.

I think now that you've re-done your RPi setup it's more likely related to https://github.com/sandeepmistry/noble/issues/242 and  https://github.com/sandeepmistry/node-sensortag/issues/55 now.

I can't see the hcidump output in your gist (https://gist.github.com/oroce/27287ae1cf36ee19cae7). However, if you check it and compare to my notes in https://github.com/sandeepmistry/noble/issues/242 we can confirm ...

oroce, 2015-09-28:
Oh indeed hcidump is missing. sorry for that.

closing this, related to #242

closed 2015-09-28 by oroce