sandeepmistry/node-sensortag #59: Accelerator - Gyroscope sensor mistaken

unicorn42, 2015-10-04:
Hi

I load the sensordata to a InfluxDB and visualize the data with Grafana. There I found the problem, that the gyroscope data are the data from the accelerator and vise versa. See the attached screenshot.

[image]

I turned the sensor in different directions (12:56:50 or 12:57:05)
At 12:57:35 i trhew the sensor in the air and at 12:57:50 i let fall it on the table.

You can check out the source here: 
https://github.com/makeplantsnotwar/node-bluetooth-connector/blob/master/app.js

Thanks

sandeepmistry, 2015-10-04:
@unicorn42 what Sensortag model are you using?

It's probably best for you to dive into the code and submit a pull request with any fixes you make. You can use the following links for reference to the data formats and formulas:

 * http://processors.wiki.ti.com/index.php/SensorTag_User_Guide
 * http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide

unicorn42, 2015-10-08:
@sandeepmistry  We used the CC2650 Sensortag.

I just found the bug while i setup Grafana. I have no clue about javascript nor node specific stuff. So i can't really dive into the code. It should be a short thing, just exchange the identifiers and you're done ;-)

sandeepmistry, 2015-10-12:
@unicorn42 alright, I double-checked the [conversion code](https://github.com/sandeepmistry/node-sensortag/blob/master/lib/cc2650.js#L177) with the [TI docs](http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#Movement_Sensor), and the calculations match.

The TI SensorTag iOS app also shows the same values. However they are probably wrong as well. Which firmware version are you running on the SensorTag? I just updated to 1.20, I wonder if they change something ....

unicorn42, 2015-10-21:
I'dont have the SensorTag anymore available to answer that question. Just change the position of de Sensor and you will see the graphs changing at the accelerator. It the tag is not moving accelerator should be 0, but it's not. Thats why its wrong.

sandeepmistry, 2015-11-26:
Some notes from Gitter:

@zywind Nov 24 11:31
> I asked about this on the TI support forum here: https://e2e.ti.com/support/wireless_connectivity/f/538/t/471879
> It turns out the sensor data are non-calibrated raw data. So I need to calibrate it in my application
> For the magnetometer, I could probably use the method described here: https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

sandeepmistry, 2015-12-27:
I'm marking this as a duplicate of #62, I've updated the accelerometer conversion formulas, as they were off by a value of 4.

Acceleration should be +/- 1g when not moving, to account for gravity.

closed 2015-12-27 by sandeepmistry