sandeepmistry/node-sensortag #60: Can't install dependency

windrago, 2015-11-04:
xpc has a cow if you don't set the 2.7 version of python which I did. I just tried to run the samples linked in the read me and it blows. Any suggestion on how to fix it?
=> node logger.js 
module.js:338
    throw err;
          ^
Error: Cannot find module 'xpc-connection'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/Users/amok/node_modules/noble/lib/mac/yosemite.js:7:21)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)

sandeepmistry, 2015-11-07:
@windrago unfortunately node-gyp only supports Python 2.7: https://github.com/nodejs/node-gyp#installation

node-xpc-connection depends on node-gyp to build the native components.

windrago, 2015-11-08:
thx - made the switch and after several hammering all works :-)

sandeepmistry, 2015-11-09:
@windrago great!

I've adding a prerequisites section to the read me: https://github.com/sandeepmistry/node-sensortag/commit/d12f5b8d808397168a377396d77fa28280214e5c

closed 2015-11-09 by sandeepmistry

windrago, 2015-11-09:
for whomever drops here and needs help in doing this switch - here is what you need to do:
npm config set python /usr/bin/python2.7
(restart terminal helps but not necessary)

sandeepmistry, 2015-11-14:
@windrago great, thanks for the follow up.