sandeepmistry/node-sensortag #62: CC2650: Wrong scale for accelerometer values

AlexDmr, 2015-12-14:
Hi, thanks for this great project!
I get wrong acelerometer values from my CC2650 sensorTag, I have to multiply values by 4 to get actual acceleration. It seems that the initialisation of scale is not correct? 
When I observe values with the Android application values are correct. 
Any idea where it could come from ?

snowbreeze, 2015-12-18:
I think we need to set the sensitivity of the accelerometer.  Change to 8G or 16GB. 
In the app, it sets it to 8G somehow.

AlexDmr, 2015-12-18:
Indeed, but it is currently not possible with node-sensortag. Should be a parameter of the sensorTag.enableAccelerometer method ?

sandeepmistry, 2015-12-20:
@AlexDmr @snowbreeze is the source code to the Android app open source? Do you have a link to it?

Which firmware version do you have on the SensorTag?

This would probably help resolve #59 as well.

Note: bit's 8-9 are used to set the accelerometer range (http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#Configuration_2)

node-sensortag is setting config to 0x7f: https://github.com/sandeepmistry/node-sensortag/blob/master/lib/cc2650.js#L108 - does this mean 16G is set?

AlexDmr, 2015-12-20:
Hi, the Android application can be downloaded from the market (sensortag Texas instrument). However i don't think that the code is open source.
The firmware installed is the one from july (don't remember the precise version), i think it is the last one as the application automatically propose an update if it is not the case.
Indeed I also see the doc you point out, I think that it could be used to specify actual ranges. As far as I understand it, 0x7f means that accéléromètre, gyroscope and magnetometer are activated. Bytes 8 and 9 are set to 0 I think, which means that scale is set to 2. This seems consistant with thé value that I observe (do not reach more than x2). So the formula used to translate values may be wrong?

sandeepmistry, 2015-12-26:
@AlexDmr I'm running firmware version```1.20 (Jul 28 2015)```, things seemed ok with using the mask ```0x007f``` on my SensorTag. However, if I changed the mask to ```0x0038``` (just accelerometer bits set), I get the 1/4th values you are seeing.

I managed to find some TI code for Android here: 
 * https://git.ti.com/sensortag-20-android/sensortag-20-android/blobs/master/sensortag20/BleSensorTag/src/main/java/com/example/ti/ble/sensortag/Sensor.java
 * https://git.ti.com/sensortag-20-android/sensortag-20-android/blobs/master/sensortag20/BleSensorTag/src/main/java/com/example/ti/ble/sensortag/SensorTagMovementProfile.java

It uses a mask of ```0x27f```, and the 8G conversion formula, so I think the accelerometer mask values currently documented are not accurate.

I've pushed some changes to master, would you mind trying them out?
```npm install sandeepmistry/node-sensortag```

AlexDmr, 2015-12-26:
Pretty good shot :)
Indeed, values are now correct when I only activate the accelerometer. Thanks a lot!

sandeepmistry, 2015-12-27:
@AlexDmr excellent, thanks for trying out the changes! I'll publish an updated version to npm shortly.

closed 2015-12-27 by sandeepmistry

tobybrad, 2016-05-12:
Hey.. I've been playing with the CC6250 on Windows. Firmware is 1.20.

I currently find it impossible to change accelerometer range. I can easily switch on and off the various gyro, accelerometer, magnetometer and WOM setting and all works as expected but bits 8:9 seem permanently set to 0x02 == 8G (Verified by directly reading the characteristic). 

Doesn't matter what I do to try and change it, range is stuck at 8G. Seems like a bug in either the firmware or the Windows API i'm using. Not a deal breaker for me but I did waste some time figuring this out.

snowbreeze, 2016-05-13:
@tobybrad Can you tell me what are you writing in place of 0x007 to turn on individual settings.
Or maybe share a bit of your code from that area.

sandeepmistry, 2016-05-15:
@tobybrad 

> Seems like a bug in either the firmware or the Windows API i'm using. 

I think it's a firmware issue on the SensorTag.