sandeepmistry/node-sensortag #64: Magnetometer cannot be enabled when accelerometer and gyro are enabled

zywind, 2016-01-04:
Seems this change broke something:
https://github.com/sandeepmistry/node-sensortag/commit/a99854261d8961fd9c4243823eb8a21cbc298ef9

When revert the flag to 0x007f, I can enable magnetometer again.

snowbreeze, 2016-01-14:
I agree

sandeepmistry, 2016-01-23:
@zywind @snowbreeze have you looked into fixing?

It feels like a firmware bug to me, and we're battling between fixing the either #62 or this behaviour.

zywind, 2016-01-23:
Is node-sensortag using a wrong mask for the accelerometer? The table:
Bits  Usage
0	Gyroscope z axis enable
1	Gyroscope y axis enable
2	Gyroscope x axis enable
3	Accelerometer z axis enable
4	Accelerometer y axis enable
5	Accelerometer x axis enable
6	Magnetometer enable (all axes)
Using the old 0x007f mask worked because it simply puts 1 on all 7 bits.  Gyro's mask is 0x0007 which puts 111 on the first three bits.  Accelerometer's mask is 0x0238.  This doesn't seem right.  Should it be 0x0038?  Magnetometer's mask is 0x0040, which looks right.

zywind, 2016-01-23:
By the way, when I revert the mask to 0x007f, I still have the 8G accelerometer range. I am not sure why, but this means that if we set the mask to 0x007f, #62 won't be an issue.

sandeepmistry, 2016-01-24:
@zywind good to know 8G range is still enabled with mask 0x007f, I am seeing the same here with f/w 1.20 (Jul 28 2015).

> Accelerometer's mask is 0x0238. This doesn't seem right. Should it be 0x0038? Magnetometer's mask is 0x0040, which looks right.

The 0x0200 was to enable 8G range, but it doesn't look like it does anything. I would say there's a bug in the firmware or misinformation in the wiki page.

Anyways, I've reverted a99854261d8961fd9c4243823eb8a21cbc298ef9, so the old style is now used (af260cfe9a6d6720e93dc1f0228f513dadefe0e0).

closed 2016-01-24 by sandeepmistry