sandeepmistry/node-sensortag #65: Cannot close onDiscover after disconnect

martin-doyle, 2016-01-06:
If a SensorTag gets disconnected the ```onDiscover``` function is still running.
In ```test.js``` of SensorTag, the script is closed via ```process.exit(0)```. This is OK, if you have only 1 SensorTag running and you want to stop the whole script in case of a disconnect.
Is it possible to close the ```onDiscover``` function when a disconnect happens?

sandeepmistry, 2016-01-08:
@martin-doyle what makes you say ```onDiscover``` is still running?

martin-doyle, 2016-01-08:
@sandeepmistry 
If I skip the ```process.exit(0)``` in the ```test.js``` than the program keeps running and needs to be stopped by ```Ctrl C```.
```javascript
  sensorTag.on('disconnect', function() {
    console.log('disconnected!');
    process.exit(0);
  });
```
In my application I do a reconnect in case of a lost connection (for example if I change the battery):
```javascript
  sensorTag.on('disconnect', function () {
    console.log('Sensortag disconnected.');
    SensorTag.discoverAll(onDiscover);
  });
```
Then, with every reconnect the disconnect message 'Sensortag disconnected.' gets fired multiple times telling me all ```onDiscover``` functions are still running. I have not found any solution in the samples other than ``` process.exit(0) ```.

sandeepmistry, 2016-01-09:
@martin-doyle for the exit question, let's move the discussion here: https://github.com/sandeepmistry/noble/issues/299

It looks like you want to use ```.once``` instead of ```.on``` as your are not removing the listener, and noble re-uses the peripheral / sensor tag instance so multiple listeners are added.

```javascript
  sensorTag.once('disconnect', function () {
    console.log('Sensortag disconnected.');
    SensorTag.discoverAll(onDiscover);
  });
```

closed 2016-01-09 by sandeepmistry

martin-doyle, 2016-01-09:
@sandeepmistry Just cool ! This is exactly I was looking for.
The noble framework works like a charm.
Thanks a lot.