sandeepmistry/node-sensortag #66: Discover sensortags

devmrko, 2016-01-27:
Hi, there
I'm trying to make IoT environment with the SensorTag. My purposes are as below.

1. Several SensorTags
2. Multiple SensorTag BLE info transmitter, I am consider the Linux machines with Bluetooth dongle.
3. Programming language is going to be NodeJS with the SensorTag NPM module.

* I know that python is another possible option, but I just don't even have any clue which one is the right choice.

4. The SensorTags could be both in and out of the range.

* It means the demon has to be always discoverable, like when the SensorTag is in the range it has to try to connect with the transmitter, and when it is out of range the demon automatically disconnects the SensorTag, and releases the resource.

Now, I am trying to make a scratch using 'node-schedule', and  'async' module.
Because I could not find the way of make infinity discover logic, I decide to use 'node-schedule' to invoke discover function with the specific interval. The problem is that discover function makes some kinds of event and it doesn't close even if the SensorTag is disconnected in the end of async series.

Thus, I want to solve those problems. Any help would be appreciated. Thanks

devmrko, 2016-01-27:
The sample code is as below.

var schedule = require('node-schedule');
var SensorTag = require('./index');
var async = require('async');
var tags = [];
var cnt = 1;
var util = require('util');

require('events').EventEmitter.prototype._maxListeners = 100;

var rule = new schedule.RecurrenceRule();
rule.second = [2, 4, 6, 8, 10, 12, 14, 16, 18, 20, 22, 24, 26, 28, 30,
32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58];

var j = schedule.scheduleJob(rule, function(){

  SensorTag.discover(function (sensorTag) {
    
    util.log('count: ' + cnt++);
    util.log('discovered:' + sensorTag);
    
    sensorTag.on('disconnect', function () {
      util.log('disconnected!');
      process.exit(0);
      var index = tags.indexOf(sensorTag.uuid);
      if(index >= 0) tags.splice(index, 1);
    });

    async.series([
      function (callback) {
        if(tags.indexOf(sensorTag.uuid) >= 0) {
          util.log('existed, sensorTag count is ' + (tags.length));
          util.log('tags: ' + tags);
          callback();
        } else {
          util.log('new one, sensorTag count is ' + (tags.length));
          util.log('tags: ' + tags);
          tags.push(sensorTag.uuid);
        }      
      },
      function (callback) {
          util.log('end of callback');
          sensorTag.disconnect(callback);
          callback();
          SensorTag.stopDiscoverAll(onDiscover);
      }
    ]);
  });
});

sandeepmistry, 2016-01-30:
@devmrko I would suggest calling ```discover``` again only after device is discovered for example:

```javascript
function sensorTagDiscovered(tag) {
  // ... some connection stuff

  discoverOne();
});

function discoverOne() {
  SensorTag.discover(sensorTagDiscovered);
}

discoverOne();
```

closed 2016-01-30 by sandeepmistry