sandeepmistry/node-sensortag #67: sensorTag.once is not documented

roestigraben, 2016-02-04:
I see this 

```
sensorTag.once(disconnect, function(){
})
```

in a couple of code example and can imagine what it does but it would be great if this is added to the documentation. Thx

sandeepmistry, 2016-02-06:
@roestigraben thanks for the feedback, I've add this in https://github.com/sandeepmistry/node-sensortag/commit/6a14fea9a1e0a9c8fe7b0343fddfdf6652baab8a. Please let me know if this change looks good.

closed 2016-02-06 by sandeepmistry

roestigraben, 2016-02-06:
> <reply+004a73cbcb5439c193bac7a30a3f55f6958ef6290e90bb8692cf0000000112cd01389
> 2a169ce07d488d5@reply.github.com>

Sandeep,

thanks a lot!

Could you have a look also at my problem with reliably connect and discover
my SensorTag under

https://gitter.im/sandeepmistry/node-sensortag

Thanks
Peter 

I've add this in 6a14fea
<https://github.com/sandeepmistry/node-sensortag/commit/6a14fea9a1e0a9c8fe7b
0343fddfdf6652baab8a> . Please let me know if this change looks good.


<https://github.com/sandeepmistry/node-sensortag/issues/67#issuecomment-1806
34782> .