sandeepmistry/node-sensortag #68: CC2650 - Read accelerometer with period 2000 does not work

johnaknight, 2016-02-23:
According to the ReadMe file, a CC2650 accelerometer can be read periodically: 

> // CC2650: period 100 - 2550 ms, default period is 1000 ms
> sensorTag.setAccelerometerPeriod(period, callback(error));

I've used a period of 2000ms and the callback is only called once with x, y, z as cero.
If I change the period to 1500ms, the callback works fine.

According to the user guide, 2000ms period should work:

> Resolution 10 ms. Range 100 ms (0x0A) to 2.55 sec (0xFF)

sandeepmistry, 2016-03-06:
@johnaknight could you please share a full code snippet that I can run to reproduce the issue?

Also, what firmware version is running on your SensorTag?

sandeepmistry, 2016-03-20:
I'm closing this due to lack of activity, please re-open when requested info if you are still interested.

closed 2016-03-20 by sandeepmistry