sandeepmistry/node-sensortag #69: Microphone & battery level support?

JonSilver, 2016-03-14:
Hi Sandeep,

Thank you for your excellent work on supporting the SensorTags in a wider community of developers. I was wondering if you're planning another release to support the CC2650's microphone and battery level data?

Kind regards,
Jon

sandeepmistry, 2016-03-20:
@JonSilver if there is documentation or examples of other apps using the microphone and battery level, sure.

However, I do not see any information on the wiki page: http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide

Are you aware of another source that does have it?

sandeepmistry, 2016-04-03:
I'm closing this due to lack of feedback, please re-open and answer my question above if you are still interested.

closed 2016-04-03 by sandeepmistry