sandeepmistry/node-sensortag #71: Raspbian boot

joseLourenco, 2016-03-24:
Hi,

How can I put my node.js app, with node_sensortag, running after the boot?

Thanks

danielbh, 2016-03-24:
I had problems with this yestarday. I tried to do it in the init.d file but supervisor causes huge memory usage issues when run there.  You might be able to do an init.d set up more easily in your project. For now we had to hack it with crontab. Would love to figure out the proper way. Additionally we had to delay it to run to work. I believe it's because Bluetooth isn't activated when crontab reboots are done initially. This is how we did it.

sudo crontab -e

@reboot sleep 10 && cd /project-directory && /path-to-node/node app.js > app.logs

Let me know if that works. Again it's a temporary hack until we figure out the proper way.

joseLourenco, 2016-03-24:
Hi,

First of all, thanks for your reply.

This "@reboot sleep 10 && cd /project-directory && sudo /path-to-node/node app.js > app.logs" is a command line right?

My project directory is /home/pi/poRsCHE.
But the "path-to-node" what is it? Is like "usr/bin/node"? How do I know the right address?

danielbh, 2016-03-24:
No you put the @reboot in the crontab config file.
'which node' to find the path

Again I want to re-emphasize this is a hack imo and to try to do it in the init.d directory instead of crontab. See above links for more info.

Also if anyone watching this repo wants to work with me to develop a better solution I'm all ears.

joseLourenco, 2016-03-24:
Ok, thanks.

I'm going to test and I'll report you.

joseLourenco, 2016-03-24:
It worked and it didn't work...

In my app I have a bleno service to receive data from my Android app and I have node-sensortag to receive data from the SensorTag. 
After I boot the Raspberry, it advertises the bleno service but after it receives the data from the Android app, it doesn't connect with the SensorTag.

Do you have som explanation for this? :S

danielbh, 2016-03-24:
I have no idea. We only use this module for bluetooth communication to the sensortag.  We have no communication directly to mobile devices with bluetooth.

danielbh, 2016-03-24:
Sorry I can't help. We will eventually implement bluetooth to mobile devices, but not at this time.

joseLourenco, 2016-03-24:
The bleno modules was developed by the same author of the node-sensortag, Sandeep Mistry.

My app works well when I execute it by command line. It's just not working well by boot (by your guide).

I'll try other guides. 
Meanwhile if you have some new idea, I'm here.

Thanks for your support.

danielbh, 2016-03-24:
Yeah there is something going on behind the scenes on boot up. I'd be very interested to know a way to make this solution portable.

joseLourenco, 2016-03-25:
Hi

I have success with the crontab file but with a line of code a bit different:
@reboot sleep 10 && su pi -c 'sudo node-path/node /home/pi/appDirectory/app.js < /dev/null &'

I hope it will help you.

Best regards

danielbh, 2016-03-25:
The last part from what I sent your writes to a log file. Not sure what your code is doing.

joseLourenco, 2016-03-26:
the problems was not you line of code with logs.

was my js app that had a process.stdout that was killing the connection with the sensortag.

sandeepmistry, 2016-03-28:
@joseLourenco I'm closing this now, as it looks like you question is solved.

In the future, please only create issues for node-sensortag related items, there are much better places to ask general questions related to Linux and Node.js.

@danielbh thank you for helping out!

closed 2016-03-28 by sandeepmistry