sandeepmistry/node-sensortag #74: Undefined return values from readings

OmarHallab, 2016-04-04:
Dear Sandeep,

I am currently facing an issue with retrieving Sensortag values. I am working on a raspberry pi 2 with node js v 4.4.2 . 

When i start my node application I am able to discover the sensortags in range, but when it comes to function callbacks i get undefined results from the readings. 

I tried tracing the application to see why the syncing is not happening: 

sensorTag.enableIrTemperature(function (error) {
            if (error)
            {
                console.log("Sensor enabling issue: %s...", error);
                return;
            } 

then i found out that : 

Sensor enabling issue: Error: service uuid f000aa0004514000b000000000000000 not found!

Can you please help me with this?

Best regards,
Omar

OmarHallab, 2016-04-06:
Dear Sandeep,

I have made sure that the devices are 100% working and that all the service UUID's are up & running. 

Though I still don't get why i can't seem to return values from the sensors, keep getting error that the certain "*************** uuid is not found! "

Any thoughts on this? 

Thanks Omar

sandeepmistry, 2016-04-09:
@OmarHallab please share the code you are running. It sounds like you are missing a call to [```sensorTag.connectAndSetUp()```](https://github.com/sandeepmistry/node-sensortag#connect-and-set-up).

Also, have you tried running [test.js](https://github.com/sandeepmistry/node-sensortag/blob/master/test.js)?

OmarHallab, 2016-04-10:
Dear Sandeep,

Test.js works fine & I am able to retreive data from the sensor. 
I have shared the code below, 

---------

var SensorTag = require('sensortag'); // sensortag library
var Async = require('async');

var Protocol = require('azure-iot-device-amqp').Amqp;
var Client = require('azure-iot-device').Client;
var ConnectionString = require('azure-iot-device').ConnectionString;
var Message = require('azure-iot-device').Message;

var connectionString = 'HostName=XXX.azure-devices.net;DeviceId=home-meteo-device;SharedAccessKey=YYY';
var deviceId = ConnectionString.parse(connectionString).DeviceId;

var client = Client.fromConnectionString(connectionString, Protocol);

SensorTag.discoverAll(function (sensorTag) {
console.log("Connecting to %s...", sensorTag.id);

   sensorTag.on('disconnect', function() {
        console.log("Disconnected from %s!", sensorTag.id);
        process.exit(0);
    });

    sensorTag.connectAndSetUp(function (error) { 
        console.log("Connected to %s...", sensorTag.id);

        sensorTag.enableIrTemperature(function (error) {
            if (error)
            {
                console.log("Sensor enabling issue: %s...", error);
                return;
            } 

            sensorTag.readIrTemperature(function (error, objectTemperature, ambientTemperature) {
                if (error)
                {
                    console.log("IR temperature error: %s...", error);
                    return;
                }

                console.log(ambientTemperature);
            }); 
        });
    });
});

----------------------------

Please get back to me on this :) 
Thank you,
Omar

sandeepmistry, 2016-05-09:
@OmarHallab which model SensorTag are you using? Does the ```connectAndSetUp``` callback arg have an error?

sandeepmistry, 2016-05-15:
This seems ok on my Raspberry Pi 3 with the CC2650 sensor tag model:

```javascript
var SensorTag = require('./index'); // sensortag library

SensorTag.discoverAll(function (sensorTag) {
    console.log("Connecting to %s...", sensorTag.id);

    sensorTag.on('disconnect', function() {
        console.log("Disconnected from %s!", sensorTag.id);
        process.exit(0);
    });

    sensorTag.connectAndSetUp(function (error) {
        console.log("Connected to %s...", sensorTag.id);

        sensorTag.enableIrTemperature(function (error) {
            if (error)
            {
                console.log("Sensor enabling issue: %s...", error);
                return;
            }

            sensorTag.readIrTemperature(function (error, objectTemperature, ambientTemperature) {
                if (error)
                {
                    console.log("IR temperature error: %s...", error);
                    return;
                }

                console.log(ambientTemperature);
            });
        });
    });
});
```

sandeepmistry, 2016-06-06:
I'm closing this due to lack of activity, please re-open with requested info if you are still interested.

closed 2016-06-06 by sandeepmistry

akshayghadage, 2019-06-17:
Hi Sandeep, 
I am facing same issue. Newly installed raspbian with Rpi Zero, Newly bought SensorTag (Rev 1.5.1, FW version- 1.42. Any suggestions??

rzumbuehl, 2020-04-13:
Hi Sandeep,

same issue here. FW version = 1.30, HW rev. = 1.2/1.3

test.js results in: 

```
discovered: {"id":"247189e73701","type":"cc2650"}
connectAndSetUp
readDeviceName
	error = Error: **characteristic uuid 2a00 not found in service uuid 1800**!
	device name = undefined
readSystemId
	system id = undefined
readSerialNumber
	serial number = undefined
readFirmwareRevision
	firmware revision = undefined
readHardwareRevision
	hardware revision = undefined
readSoftwareRevision
	software revision = undefined
readManufacturerName
	manufacturer name = undefined
enableIrTemperature
(node:3516) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): **Error: service uuid f000aa0004514000b000000000000000 not found**!
```

Any ideas?