sandeepmistry/node-sensortag #76: Read all sensors at once

meeximum, 2016-04-26:
Hi,

what would be the best approach to read all values at once?
I tried with async.parallel() but it doesn't worked :-/

regards
Meex

sandeepmistry, 2016-05-01:
@meeximum could you please share what you've tried so far?

Maybe something like [test.js](https://github.com/sandeepmistry/node-sensortag/blob/master/test.js) but to enable notifications on each sensor instead and listen for events.

meeximum, 2016-05-02:
@sandeepmistry ok when I do like in test.js, I just need to comment out the setTimeout(function() {} to make it run forever?

sandeepmistry, 2016-05-08:
@meeximum see something like this: https://github.com/tigoe/BluetoothLE-Examples/blob/master/sensorTag/sensorTagExample/sensortagIRTemperature.js

closed 2016-05-08 by sandeepmistry