sandeepmistry/node-sensortag #77: 16G Accelerometer

snowbreeze, 2016-04-29:
Hi @sandeepmistry,
Line 184 lib/cc2650.js it says // we specify 8G range in setup.
How do i change to 16g or 4g.

sandeepmistry, 2016-05-02:
This is currently not supported.

However can be, by looking adding a API to set the correct bits: http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#Configuration_2

Pull requests welcome ....

openStove, 2016-07-27:
Hi sandeepmistry,

I'm not sure to understand, is it possible to set the bits somewhere manually in your code (replace a mask?), or where those bits are to be setup ?

sandeepmistry, 2016-07-28:
@openStove try https://github.com/sandeepmistry/node-sensortag/blob/master/lib/cc2650.js#L19 and the link I mentioned earlier. All the code you need to change is isolated in `lib/cc2650.js`. Note: the SensorTag firmware might have issues, so the docs don't always line up.

efreiberg, 2016-09-14:
@sandeepmistry Is it enabling 8G by default on the accelerometer?  Looking at https://github.com/sandeepmistry/node-sensortag/blob/master/lib/cc2650.js#L108 it seems it is being set to 2G by using 0x007f for the config characteristic value?

sandeepmistry, 2016-09-17:
Hi @efreiberg,

> it is being set to 2G by using 0x007f for the config characteristic value?

Maybe, from what I remember there was some bugs in the firmware or documentation when I was looking at it. Maybe this has since been fixed, the `0x007f` was the only way I could get things working reliable.

Please tryout other values and report your findings (also mention the f/w version you are using on your SensorTag). Thanks.

abhatikar, 2017-06-29:
Hi Sandeep, I am a novice in nodejs.I had a question related to Ti Sensor tag reading MPU9250 data
i am trying to understand how to read motion sensor data using gatttool
I see in your code you see the config mask as 0x7F which is enabling all axes for all 3 sensors/meters
when I try to do this using gatttool, i am not able get any data from the handle associated with AA81*
Could you please help

abhatikar, 2017-06-29:
abhatikar@abhatikar-mac:~$ gatttool -I -b 24:XX:89:XX:XX:XX
[24:XX:89:XX:XX:XX][LE]> connect
Attempting to connect to 24:XX:89:XX:XX:XX
Connection successful
[24:XX:89:XX:XX:XX][LE]> char-read-uuid f000aa82-0451-4000-b000-000000000000
handle: 0x003f value: 00 02
[24:XX:89:XX:XX:XX][LE]> char-write-req 0x3f 007f
Characteristic value was written successfully
Notification handle = 0x003c value: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[24:XX:89:XX:XX:XX][LE]> char-read-uuid f000aa81-0451-4000-b000-000000000000
handle: 0x003c value: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[24:XX:89:XX:XX:XX][LE]>
Please find the logs for your reference

abhatikar, 2017-06-29:
My Firmware revision:**1.32 Oct 5 2016** 
Model: **CC2650 Sensor tag**