sandeepmistry/node-sensortag #80: Problem using node-sensortag with noble

AlexDmr, 2016-06-23:
I am using both node-sensortag and noble in my code, the problem is that when I require both I get a : 
Error: Could not initialize libusb. Check that your system has a usb controller.

if I require only one of them it works. Here are the versions I am using :  
     "noble": "^1.6.0",
    "sensortag": "^1.2.2",

both installed via npm install --save ...

sandeepmistry, 2016-06-27:
@AlexDmr please share the following:

* A minimal code snippet to reproduce
* How you are launching the application. From my understanding libusb is used when some env var's are set
* Full stack trace of error.

AlexDmr, 2016-06-27:
Hi, I am using windows 10, with node 4.4.5.
The trace error : 
C:\Recherche\ErrorNoble\node_modules\sensortag\node_modules\noble-device\node_modules\noble\node_modules\bluetooth-hci-socket\node_modules\usb\usb.js:11
        throw new Error('Could not initialize libusb. Check that your system has a usb controller.');
        ^

Error: Could not initialize libusb. Check that your system has a usb controller.
    at Object.<anonymous> (C:\Recherche\ErrorNoble\node_modules\sensortag\node_modules\noble-device\node_modules\noble\node_modules\bluetooth-hci-socket\node_modules\usb\usb.js:11:8)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (C:\Recherche\ErrorNoble\node_modules\sensortag\node_modules\noble-device\node_modules\noble\node_modules\bluetooth-hci-socket\lib\usb.js:5:11)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)

----- main.js
var noble		= require( "noble" )
  , sensortag	= require( "sensortag" )
  ;

noble.on('scanStart', function() {console.log("scanStart");} );

----- NPM : 
{
  "name": "errornoble",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "author": "",
  "license": "ISC",
  "dependencies": {
    "noble": "^1.6.0",
    "sensortag": "^1.2.2"
  }
}

AlexDmr, 2016-06-27:
Sorry, forgot to mention that I am launching the app with : 
node main.js

sandeepmistry, 2016-07-03:
@AlexDmr thanks, now that you've mentioned you're using Windows this makes more sense :)

This is a downside of using noble on Windows now, each instance of noble needs to open the USB device directly.

What's the output of `npm ls`? If `node-sensortag` is not using noble `v1.6.0`, can you try changing it to do so? (By removing `node_modules/sensortag/node_modules/noble` and running `npm install noble@1.6.0` in `node_modules/sensortag/node_modules`)

AlexDmr, 2016-07-03:
Indeed ! sensorTag installed noble 1.3.0 ! Why ???
Ho actually it is not directly sensorTag but noble-device that install noble 1.3.0
still... why? :)

sandeepmistry, 2016-07-04:
This is why: https://github.com/sandeepmistry/noble-device/blob/master/package.json#L34

It can probably be changed from "~1.3.0" to "^1.3.0" ...

sandeepmistry, 2016-08-22:
Closing, this change has been made upstream: https://github.com/sandeepmistry/noble-device/commit/22fadf929cbb58c526bffa497014d1956546ac39#diff-b9cfc7f2cdf78a7f4b91a753d10865a2

closed 2016-08-22 by sandeepmistry