sandeepmistry/node-sensortag #84: Raspberry pi2 + node-sensortag

julianoAffonso, 2016-08-22:
I followed strictly install process about node-gyp and noble before install this package in my project, after some minutes, finally I have success (or I think have sucess) to install noble (don't received any fail noble instalation and node-gy rebuild process).

I try run test.js from node-sensortag and not receive any error and any information.

I'm able to make communication from my Raspberry Pi 2 with my dongle generic Bluetooth and write and read information from my TI sensortag with Bluez normally.

I tried to get services and characteristics using just noble and cannot get this information, I'm able to get just information from peripherals (ID, MAC, name etc).

Someone have a similar issue like that?

sandeepmistry, 2016-08-24:
@julianoAffonso are you running using `sudo`? This might not be in the read me ... let me know.

julianoAffonso, 2016-08-24:
@sandeepmistry Thanks for your response, yes, I'm running using `sudo`.

sandeepmistry, 2016-08-24:
ok, let's take a look at the following:

* debug output: `sudo DEBUG=* node test.js`
* HCI dump: `sudo hcidump -t -x`

julianoAffonso, 2016-08-24:
```
  sudo DEBUG=* node test.js
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +45ms
  hci set le event mask - writing: 010120081f00000000000000 +18ms
  hci read local version - writing: 01011000 +2ms
  hci write LE host supported - writing: 016d0c020100 +3ms
  hci read LE host supported - writing: 016c0c00 +2ms
  hci read bd addr - writing: 01091000 +2ms
  hci onSocketData: 040e0401010c00 +101ms
  hci 	event type = 4 +3ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 3073 +2ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  hci onSocketData: 040e0401012000 +4ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 8193 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  hci onSocketData: 040e0c0101100006bb22060a00bb22 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 4097 +0ms
  hci 		status = 0 +0ms
  hci 		result = 06bb22060a00bb22 +1ms
  hci set scan enabled - writing: 010c20020001 +3ms
  hci set scan parameters - writing: 010b200701100010000000 +2ms
  hci onSocketData: 040e04016d0c00 +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 3181 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e06016c0c000100 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 3180 +1ms
  hci 		status = 0 +0ms
  hci 		result = 0100 +0ms
  hci 			le = 1 +1ms
  hci 			simul = 0 +1ms
  hci onSocketData: 040e0a010910001371da7d1a00 +0ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4105 +0ms
  hci 		status = 0 +1ms
  hci 		result = 1371da7d1a00 +0ms
  hci address = 00:1a:7d:da:71:13 +5ms
  noble addressChange 00:1a:7d:da:71:13 +1ms
  hci onSocketData: 040e04010c200c +6ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +1ms
  hci onSocketData: 040e04010b2000 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble stateChange poweredOn +2ms
  hci set scan enabled - writing: 010c20020101 +6ms
  hci set scan enabled - writing: 010c20020101 +2ms
  hci onSocketData: 040e04010c2000 +15ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 8204 +0ms
[... 65 lines cut ...]
  hci 			address type = public +1ms
  hci 			eir = 020106030280aa06ff0d00030000 +1ms
  hci 			rssi = -81 +0ms
  gap advertisement = {"localName":"SensorTag 000001","txPowerLevel":0,"manufacturerData":{"type":"Buffer","data":[13,0,3,0,0]},"serviceData":[],"serviceUuids":["aa80"]} +2ms
  hci onSocketData: 043e27020104000082b6f8e6a01b110947756974617220455648205374726970051208002003020a00af +6ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +1ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event status = 1 +1ms
  hci 		LE meta event data = 04000082b6f8e6a01b110947756974617220455648205374726970051208002003020a00af +0ms
  hci 			type = 4 +1ms
  hci 			address = a0:e6:f8:b6:82:00 +0ms
  hci 			address type = public +1ms
  hci 			eir = 110947756974617220455648205374726970051208002003020a00 +1ms
  hci 			rssi = -81 +0ms
```
HCIDUMP

```
sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
```

Thanks for you help!

sandeepmistry, 2016-08-28:
Hi @julianoAffonso,

Thanks for the info. May I ask which model of SensorTag you have and which firmware is loaded? It's advertising `SensorTag 000001` as the local name, currently the following are checked:

 * CC2540: https://github.com/sandeepmistry/node-sensortag/blob/master/lib/cc2540.js#L41-L46
 * CC2650: https://github.com/sandeepmistry/node-sensortag/blob/master/lib/cc2650.js#L41-L46

We'll have to make the relevant changes to one of the files to white list what your tag is advertising.

julianoAffonso, 2016-08-29:
Hi, Thanks for your response,

I changed the name off my SensorTag using the dev pack from TI. The model is CC2650 and firmware verison is 1.21

sandeepmistry, 2016-09-05:
Hi @julianoAffonso,

Ok, I didn't know that was possible :)

I think we can change: https://github.com/sandeepmistry/node-sensortag/blob/master/lib/cc2650.js#L41-L46

to include an `||` check for `peripheral.advertisement.serviceUuids` containing `aa80`.

Could you please try this out and submit a pull request?

julianoAffonso, 2016-09-05:
Hi @sandeepmistry, thanks for your response!

I think this works, see my terminal output:

`node_modules/sensortag $ sudo node test
discovered: {"id":"a0e6f8b68200","type":"cc2650"}
connectAndSetUp
readDeviceName
	device name = SensorTag 2.0
readSystemId
	system id = a0:e6:f8:00:00:b6:82:00
readSerialNumber
	serial number = N.A.
readFirmwareRevision
	firmware revision = 1.20 (Jul  9 2016)
readHardwareRevision
	hardware revision = PCB 1.2/1.3
readSoftwareRevision
	software revision = PCB 1.2/1.3
readManufacturerName
	manufacturer name = Texas Instruments
enableIrTemperature
readIrTemperature
	object temperature = 17.3 °C
	ambient temperature = 24.4 °C
disableIrTemperature
enableAccelerometer
readAccelerometer
	x = 0 G
	y = 0 G
	z = -1 G
disableAccelerometer
enableHumidity
readHumidity
	temperature = 24.6 °C
	humidity = 42.6 %
disableHumidity
enableMagnetometer
readMagnetometer
	x = 27.3 μT
	y = 57.1 μT
	z = -79.4 μT
disableMagnetometer
enableBarometricPressure
readBarometricPressure
	pressure = 930 mBar
disableBarometricPressure
enableGyroscope
readGyroscope
	x = -2.4 °/s
	y = 1.5 °/s
	z = -1.7 °/s
disableGyroscope
readIoData
	data = 127
writeIoData
readIoConfig
	config = 0
writeIoConfig
enableLuxometer
readLuxometer
	lux = 0
disableLuxometer
readSimpleRead - waiting for button press ...`

sandeepmistry, 2016-09-05:
@julianoAffonso looks great!

Please go ahead and submit a pull request with the changes you've made.

julianoAffonso, 2016-09-05:
@sandeepmistry ,

Thanks for your support!

closed 2016-09-05 by julianoAffonso

sandeepmistry, 2016-09-06:
@julianoAffonso you're welcome, are you planning on opening a pull request?

julianoAffonso, 2016-09-06:
@sandeepmistry ,

Done!