sandeepmistry/node-sensortag #89: discoverAll() not defined error

roccomuso, 2016-09-30:
I'm trying to execute a script using this library.
But when I call <code>sensortag.discover(...);</code> I get an error like:

```javascript
TypeError: Object function () { [hidden code] }
 has no method 'discoverAll' 
```
raised on this line: https://github.com/sandeepmistry/node-sensortag/blob/master/lib/sensortag.js#L8

Do <code>cc2540.js</code> expose somewhere the `discoverAll` method?

sandeepmistry, 2016-10-02:
Hi @roccomuso,

Things seem fine to me, could you please share a script that reproduces the error?

`discoverAll` is provided by [noble-device](https://github.com/sandeepmistry/noble-device).

roccomuso, 2016-10-02:
@sandeepmistry actually it was my fault... my "environment" doesn't support that kind of class inheritance. I forgot to close the issue.

closed 2016-10-02 by roccomuso