sandeepmistry/node-sensortag #90: Check battery level on CC2650

craigmw, 2016-10-06:
TI has recently provided a method to read the battery level on the SensorTag. Is there any way to read this value using your node-sensortag library?

sandeepmistry, 2016-10-10:
@craigmw please see pull request #82.

It's not ready to merge yet, but if you can try out my suggestion in https://github.com/sandeepmistry/node-sensortag/pull/82#issuecomment-241288978 and submit a new pull request if it works, that would be great.

craigmw, 2016-10-14:
Thanks Sandeep. I'll try this out and report back.

sandeepmistry, 2016-10-30:
@craigmw have you made any progress on this?

sandeepmistry, 2016-12-05:
@craigmw ping ...

danielbh, 2016-12-29:
It looks like this is available with 1.3 revision! Check out the dev app!

[image]

sandeepmistry, 2017-01-16:
Fixed via #93.

closed 2017-01-16 by sandeepmistry