sandeepmistry/node-sensortag #91: About supporting voice over GATT/BLE for cc2650 sensortag

jarvis-hal, 2016-12-25:
It seems like the latest BLE stack from TI has code built in for using the MEMS microphone sensors on board for cc2650stk sensor tag. I have been looking in it but find it beyond my expertise to integrate it into node-sensortag. Any thoughts on adding support for voice-over-BLE?

sandeepmistry, 2017-01-03:
@jarvis-hal does the latest official TI SensorTag firmware contain this feature?

I don't see any documentation for it: http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#Gatt_Server

obriensoftwarehouse, 2017-01-03:
@sandeepmistry Do these links assist? The PDM Driver was included as part of the BLE Stack 2.2.1

https://github.com/ti-simplelink/ble_examples/blob/master/docs/simple_central_audio_receiver.md

**Additional References**
[PDM Driver](http://software-dl.ti.com/dsps/dsps_public_sw/sdo_sb/targetcontent/mcusdk/2_20_01_08/exports/tirtos_full_2_20_01_08/products/tidrivers_full_2_20_01_10/docs/doxygen/html/_p_d_m_c_c26_x_x_8h.html)
http://processors.wiki.ti.com/index.php/CC2650RC_Getting_Started_with_Development#Voice_Streaming_on_the_CC2650RC
http://processors.wiki.ti.com/index.php/BLESDK-2.2.x-CC2650RC_Developers_Guide#Voice_Over_BLE
[Voice Over Remote Control](http://www.ti.com/lit/an/swra506/swra506.pdf)

sandeepmistry, 2017-01-16:
@obriensoftwarehouse does the stock SensorTag firmware have these features bundled, or is a custom build required?

jarvis-hal, 2017-01-19:
I believe that the stock firmware does not have the voice built-in. I will try to test and compile the firmware coming with the Bluetooth SDK 2.2.1 and see how to enable this.

sandeepmistry, 2017-02-06:
@jarvis-hal thanks for the info.

For now, I think I will close this issue/request, since I expect a majority of users will probably only use the stock firmware on their SensorTags. So, it's best to wait for TI to include this feature.

closed 2017-02-06 by sandeepmistry

jarvis-hal, 2017-02-06:
Yes, it will be interesting to see when TI is going to release the voice-over-BLE as a stock firmware. I will try to play around the build and give some updates when I make some progress.

santiag0aragon, 2017-03-14:
@sandeepmistry @jarvis-hal To enable voice-over-GaTT on the sensortag is enough to compile and install in the sensortag the sensortag-audio_app + sensortag-audio_stack examples provided in the SDK.