sandeepmistry/node-sensortag #92: Big difference in power consumption when using this lib with iMac versus rpi3 raspbian

danielbh, 2016-12-28:
Hello!

First I used this script: https://github.com/danielbh/sensortag-battery-test/blob/master/polling.js
Then I followed this guide: http://mobilemodding.info/2015/06/ti-sensortag-2-power-consumption-analysys/

I tested this script on a iMac I bought this year and rpi3 from last year.

On the iMac when connecting and polling the sensors I'm getting: ~0.01 mA -0.20 mA and it looks like the iMac is actively optimizing the bluetooth connection to preserve the battery life of the sensor because the mA is rarely over 0.15.

However on the raspberry pi something quite different is occurring. I'm getting no less than 0.50 mA and it is staying consistently above that, even when  the sensors are not being read.

Any idea what is going on or where I could begin investigating this problem? Thanks for any help you could offer.

danielbh, 2016-12-29:
Update when connecting to a sensor using bluetoothctl only I'm gettting 0.09 mA - 0.18 mA power consumption

danielbh, 2016-12-29:
updated above comment with steps to repeat.

danielbh, 2016-12-29:
Made a quick and dirty test with noble directly. Just keep restarting script until you connect to the sensortag. The result was the same as the sensortag script above. ~0.5 mA. 

https://github.com/danielbh/sensortag-battery-test/blob/master/noble-connection-tester.js

What are your thoughts on this? Thanks.

danielbh, 2016-12-29:
Updating the firmware to 1.3 seems to make this issue worse. It runs then at ~0.6 mA!

sandeepmistry, 2017-01-16:
@danielbh please provide an HCI dump capture: `sudo hcidump -t -x`.

How are you measuring the power consumption?