sandeepmistry/node-sensortag #94: How to get the battery level?

rei-vilo, 2017-01-16:
[added battery service mixin from noble](https://github.com/sandeepmistry/node-sensortag/commit/3e9a664db167413e599b9f679e1ac84b5a29e423) has just been added, but how to get the battery level?

Thank you!

micahnyc, 2017-01-17:
Hi @rei-vilo you can read the battery in a similar way to the other properties:

tag.readBatteryLevel(function(error, batteryLevel) {
     console.log(tag.id, " battery read with error:", error, " level:", batteryLevel);
});

sandeepmistry, 2017-01-30:
@micahnyc thank's for helping out!

I've added it to the read me in https://github.com/sandeepmistry/node-sensortag/commit/27d820558c535a06aa8a78fd6607a026b2579292.

closed 2017-01-30 by sandeepmistry

abataille, 2017-02-08:
I tried this, but get:
 tag.readBatteryLevel(function(error, batteryLevel) {
            ^
TypeError: tag.readBatteryLevel is not a function
I cannot find it in version 1.2.3.

sandeepmistry, 2017-02-13:
v1.3.0 has been published to npm with the new feature.