sandeepmistry/node-sensortag #95: read battery level

abataille, 2017-02-09:
I tried this, but get:
tag.readBatteryLevel(function(error, batteryLevel) {
^
TypeError: tag.readBatteryLevel is not a function
I cannot find it in version 1.2.3. 
If you have, it would be nice to a new version.

micahnyc, 2017-02-09:
@abataille since this feature was added w/o a updated version number, perhaps you just don't have the updated code in your project. Try deleting the node-sensortag folder from your node_modules/ folder and then run npm install again.

@sandeepmistry perhaps a 1.2.4 version is a good idea.

sandeepmistry, 2017-02-13:
@abataille @micahnyc I've just published v1.3.0 to npm with the new feature.

@micahnyc thanks for the reminder!

closed 2017-02-13 by sandeepmistry