sandeepmistry/node-sensortag #96: [CC2541] Sensortag randomly disconnects

JoeyJo0, 2017-03-05:
My CC2541 Sensortag randomly disconnects sometimes. This happens after multiple hours of uninterrupted usage(My record is 30), but when it happens, it recurs.

See graph:
[image]

My application is designed to reconnect after approx. 30 seconds. Once is disconnects once It takes multiple tries before a proper connection can be made, but it takes a minute more to throw errors again. Every disconnect throws an error "noble warning: unknown peripheral 34b1f7d555b8"

Manually pressing the advertise button seems to solve the issue for a while.

Hcidump is included.
[hcidump.txt](https://github.com/sandeepmistry/node-sensortag/files/819887/hcidump.txt)

I'm sorry if this turns out not to be an issue with the software, but I can't find a lot of information about this online.

sandeepmistry, 2017-10-02:
@JoeyJo0 what Bluetooth adapter are you using?

fommike, 2017-10-24:
@sandeepmistry  hello, I am also experiencing the same disconnect problem with CC2650 tag when I use nodej with Raspberry Pi3. I am using the built in BCM43438 chip for Bluetooth connection with bluez-5.43 which is currently the default version of the stack. Since I have several tags and Pis, I noticed that disconnect events occur absolutely random, i.e. the tag may disconnect after a few hours or after a few days. 

The disconnect event occurs both with the default firmware version on the SensorTag, as well as with my customized version. From the nodejs side I am setting the measuring period of the sensors (apart from the temperature sensor) to 100 ms. That is the only difference with the provided test.js example. 

Any thoughts what can cause the disconnect events or how to debug such a problem? 

P.S. I am also not sure which side causes the disconnect event: Pi with nodejs running  or SensorTag.

JoeyJo0, 2017-10-24:
@sandeepmistry I am also using the Raspberry Pi built-in unit.

I think this might be a battery problem. I later noticed the errors seem to go away with a fresh battery. This might also be your problem, @fommike.

I think further experimenting might be required.

fommike, 2017-10-24:
@JoeyJo0  thanks for the comment. But in my case I connect a SensorTag directly to Pi over USB with the dev Pack. So the battery and power consumption should not be an issue here. Do you have similar setting or you power the tag from a battery only?

JoeyJo0, 2017-10-24:
@fommike As I own the CC2541 tag, I don't have the luxury to power from USB.. Good to know it's not the battery, then. It might actually be a problem with the RPi unit, then.