sandeepmistry/node-sensortag #98: Unable to update to V1.3 to read Battery Level

Clause16, 2017-03-08:
Hi guys,

I just started using RPI with SensorTag a few days back and I´m trying to read the battery level level of the sensortag. When I used npm install sensortag in the node_nodule directory, V1.3 of the code gets installed , but when node red is started, there´s no sensortag node found. To get the sensortag node to show up in Node Red, I have to install the sensortag node using npm install node-red-node-sensortag. However this command only installs V1.2.3. of the code, and reading of battery level is not supported. 

I´ve also tried to manually overwrite the sensortag folder in the node-red-node-sensortag/node_modules with the sensortag folder installed in the node_modules directory with the npm install sensortag command but this does not seem to have any effect.

Am I doing anything wrong which results in node red not picking up the V1.3 installed with npm install sensortag? Or is there anyway for me to update the V1.2.3 to V1.3 manually? 

Thanks in advance for your help!

sandeepmistry, 2017-03-13:
@Clause16 there's probably a package.json file in the directory that pins the version to v1.2.3.

Some things you can try:
1) find and edit the package.json then run the npm install again
2) run the npm install in another test directory, outside the node-red directory.

closed 2017-03-13 by sandeepmistry

Clause16, 2017-03-13:
Unfortunately, I am unable to solve the problem.

1. Are you referring to the package.json file in the sensortag directory? To my knowledge, I did not edit any of the files downloading using the npm install command. By ¨npm install¨ are you referring to npm install sensortag or npm install node-red-node-sensortag? 

2. I tried to run both the install commands in another directory and the results are the same, npm install sensortag installs V1.3 while npm install node-red-node-sensortag installs V1.2.3.

I think that a key point that you did not address in the question is that no sensortag node shows up when V1.3 is installed using npm install sensortag (sensorTag is considered a missing type). The sensortag node only shows up when sensortag V1.2.3 is installed using npm install node-red-node-sensortag. Is there something that´s preventing node red from recognizing sensortag V1.3 that´s installed using npm install sensortag?

sandeepmistry, 2017-03-13:
Please see: https://github.com/node-red/node-red-nodes/blob/master/hardware/sensorTag/package.json#L12

`node-red-node-sensortag` pins the `node-sensortag` dependency to 1.2.x.

Clause16, 2017-03-14:
Thanks for the prompt response. I´ve actually noticed this too and tried to change the dependency to 1.3.0, but that does not seem to change anything. What did you mean by editing the package.json and running npm install again??

Clause16, 2017-03-20:
Still no luck trying to figure this out.