sandeepmistry/node-sensortag #102: Unable to install bluetooth-hci-socket

red9ueen, 2017-05-13:
I'm engaged in programming a Sensortag App in Node.js with VS2015.
At first, I was trying to install bluetooth-hci-socket but failed.

Could you give me a clue to fix it, please?
Thank you in advance.

Error messages are following.

`====Executing command 'npm install bluetooth-hci-socket --save '====

npm WARN package.json nodejs-web-server1@0.0.0 No repository field.
npm WARN package.json nodejs-web-server1@0.0.0 No README data
> usb@1.2.0 install C:\Users\Dai\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\node_modules\bluetooth-hci-socket\node_modules\usb
> node-pre-gyp install --fallback-to-build
[usb] Success: "C:\Users\Dai\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\node_modules\bluetooth-hci-socket\node_modules\usb\src\binding\usb_bindings.node" is installed via remote
> bluetooth-hci-socket@0.5.1 install C:\Users\Dai\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\node_modules\bluetooth-hci-socket
> node-gyp rebuild
C:\Users\Dai\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\node_modules\bluetooth-hci-socket>if not defined npm_config_node_gyp (node "C:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (rebuild) 
Traceback (most recent call last):
  File "C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\gyp\gyp_main.py", line 18, in <module>
    sys.exit(gyp.script_main())
  File "C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 534, in script_main
    return main(sys.argv[1:])
  File "C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 527, in main
    return gyp_main(args)
  File "C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 503, in gyp_main
    options.circular_check)
  File "C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 98, in Load
    generator.CalculateVariables(default_variables, params)
  File "C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 1867, in CalculateVariables
    generator_flags.get('msvs_version', 'auto'))
  File "C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\MSVSVersion.py", line 402, in SelectVisualStudioVersion
    versions = _DetectVisualStudioVersions(version_map[version], 'e' in version)
KeyError: '2015'
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:343:16)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Windows_NT 6.3.9600
gyp ERR! command "node" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Dai\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\node_modules\bluetooth-hci-socket
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm ERR! Windows_NT 6.3.9600
npm ERR! argv "C:\\Program Files (x86)\\nodejs\\\\node.exe" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "bluetooth-hci-socket" "--save"
npm ERR! node v0.12.2
npm ERR! npm  v2.7.4
npm ERR! code ELIFECYCLE
npm ERR! bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-hci-socket@0.5.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the bluetooth-hci-socket package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls bluetooth-hci-socket
npm ERR! There is likely additional logging output above.
npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Dai\Documents\Visual Studio 2015\Projects\NodejsWebServer1\NodejsWebServer1\npm-debug.log

====npm command completed with exit code 1====
`

sandeepmistry, 2017-05-15:
@red9ueen could you please try to upgrade your Node.js and/or npm versions to a later version.

I think it related to https://github.com/nodejs/node-gyp/issues/629

red9ueen, 2017-05-15:
Thank you a lot for your advice.
Updating Node.js to v7.0.0-pre8 worked!

closed 2017-05-15 by red9ueen