sandeepmistry/node-sensortag #106: [Question] how to calibrate the magnetometer to remove gyro & accelero drift ?

gotgot1995, 2018-02-06:
Hi everyone,

I'm struggling to get smooth data from the motion sensors even after filtering the raw data with an [A.H.R.S](https://github.com/psiphi75/ahrs) .

Any ideas on how to calibrate the magnetometer ?

Perhaps you could implement a "calibrate" function.

Thanks in advance,

sandeepmistry, 2018-02-11:
The only info i can find is hear: http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#Calibration

Maybe TI's iOS or Android apps can calibrate the sensors?

> Perhaps you could implement a "calibrate" function.

Pull requests welcome for this.

closed 2018-02-11 by sandeepmistry

gotgot1995, 2018-02-12:
Hi thanks for your quick answer.

I've done some reading of the [MPU-9250 specification](http://www.invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf) and there seems to be a integrated component (DMP) to compute all three sensors data into valuable data (quaternion, pedometer, orientation etc...).

Arduino implementation exists [here](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library).