sandeepmistry/node-sensortag #107: No connection with Noble version 1.9

martin-doyle, 2018-02-18:
After reinstalling node-sensortag, I could not connect to the TI-Sensortag anymore. The script hangs at the connectAndSetup function. Since node-sensortag (1.3) and noble-device (1.41) have not changed, the change on noble to 1.9 seems to be the problem. After forcing version 1.81, everything works fine again.

curana, 2018-02-25:
I can confirm that
a) on my Macbook (High Sierra) it works with noble 1.9
b) on my Raspberry Pi 3 (current Raspbian) it does not work with noble 1.9. Downgrading to noble 1.8.1 fixes the connection issues.

With the RPi3 noble/sensortag hangs after connecting to the CC2650. It won't succeed in activating the sensors.

AzureDataGeek, 2018-03-08:
Confirm as well!  After hours of trying to figure out why a demo wasn't working.  :-)

sandeepmistry, 2018-03-12:
This should be fixed with https://github.com/noble/noble/pull/769. Please provide your feedback on that pull request.

piersbarrios, 2018-03-13:
Thank you very much! I have been strugling for three days before finding this was the issue. 

Works perfectly on RaspberryPi 3 (tell me if you want more details)

martin-doyle, 2018-03-15:
Yes, it works with version 1.9 and https://github.com/noble/noble/pull/769 on a Raspberry.
Thanks a lot.

sandeepmistry, 2018-03-19:
noble v1.9.1 has been published with a fix for this.

closed 2018-03-19 by sandeepmistry